EmuELEC/EmuELEC #144: T95 Max Plus Won't Boot into EmuElec

gcdoug, 2020-03-22:
I bought the same TV box that ETA Prime did a video about on YouTube. It is supposed to have the S905X3 chip in it. I followed his video to install onto a microSD card.  renamed the appropriate file from the device_trees folder. However, with everything I have tried, I cannot get the device to boot into EmuElec. Any suggestions? Thanks.

shantigilbert, 2020-03-22:
https://github.com/EmuELEC/EmuELEC/issues/129

closed 2020-03-22 by shantigilbert

gcdoug, 2020-03-22:
I didn't see this post and it seemed to help because I didn't know where the reset button inside the headphone jack but that seemed to help. Thanks

NewAgeRipper, 2020-03-27:
Solved mine by installing Terminal Emulator in Android and typing "reboot update" without quotes.