EmuELEC/EmuELEC #145: Screen overscan

ducdreamer, 2020-03-23:
I have EmuElec running el on my system.
The only problem is the output from emulation station is to big for my screen 1080P
the resolution in EmuELEC is also set at 1080P
This overscan also happens on the emulators.
I think it had something to do with overscan but I cannot find a setting where I can change it.
Anybody A tip?

BaltasarParreira, 2020-03-23:
I had the same problem on my old plasma TV just does 1080p@24hz, the solution is on the script setres.sh, but you have to trial and test the parameters.

This is what I had to do in my case:
1-  Test individual the parameters until you see correct the screen size, my case was this ones:
        fbset -fb /dev/fb0 -g 1920 1080 1920 2160 32 --> this should be the default for a 1080p TV
        echo 0 0 1919 1079 > /sys/class/graphics/fb0/free_scale_axis  --> also this should be the default for a 1080p TV
        echo 45 25 1870 1055 > /sys/class/graphics/fb0/window_axis  --> this the parameters you have to tweek until you have the screen inside without oversacan
        echo 0x10001 > /sys/class/graphics/fb0/free_scale  --> will set/saves the previous so you can see the changes on the TV

2- After you have the right values for you issue this command:
`cat /sys/class/display/mode`
it gives you a value that you going to use on next step

3- edit the setres.sh script like this:
`vi /storage/.config/emuelec/scripts/setres.sh`
and had in the end your configuration like this if the value retorned by the previous step doesn't exist, in my case I added this between the line with last ;; and esac  (note that the 32 value from the first command is replacede by the $BPP var):
```
"<your value from step 2 here>")                                                                  
        fbset -fb /dev/fb0 -g 1920 1080 1920 2160 $BPP                        
        echo 0 0 1919 1079 > /sys/class/graphics/fb0/free_scale_axis        
        echo 45 25 1870 1055 > /sys/class/graphics/fb0/window_axis  
        echo 0x10001 > /sys/class/graphics/fb0/free_scale           
        ;; 
```
if the value returned from step 2 already is one that exists inside the script just replace all the commands with the new ones in between the:
```
"<existing resolution>") 
....
        ;; 
```

4- issue the following after save all changes from step 3:
`setres.sh 32`
 
in my case I had to comment also these lines on the script since the file /ee_s905 never exists on my system so the script doesn't do nothing:
if [ -e /ee_s905 ]; then
....
fi --> This is the last fi on the script.

to comment just put # in the beginning of the lines, like this:
```
if [ -e /ee_s905 ]; then
....
fi
```

Hope with this you can solve you screen problem as I've been there a few days also fighting with the same annoying problem. ; )

ducdreamer, 2020-03-24:
Hello Baltasar Parreira,
This solution works great thanks for the tips.
I followed the steps 1, 2 and 3 step 4 throws up an error
But my system works great now no more over scan
I do not pretend to understand every command but when you follow the steps it works great
Thanks for the assistance
Greetings

closed 2020-03-24 by ducdreamer

chenteparao, 2020-05-26:
> I have EmuElec running el on my system.
> The only problem is the output from emulation station is to big for my screen 1080P
> the resolution in EmuELEC is also set at 1080P
> This overscan also happens [...]

Has you this problem?

zatarra76, 2021-12-14:
Hello, I know that this post in quite old but I have the same issue on my X96 Mini running Emuelec 4.3. I can't fix the overscan in the TV and I wanted to apply this tweak but I'm not able to find the setres.sh script, the scripts drawer is empty. Do you have any suggestion? Thanks

sdavides, 2022-01-07:
i have emuelec on s905w generic and i added the commands fbset/echo in setres.sh, at startup it works but if i start retroarch(game) and back in emuelec it returns as before.
I also have emuelec s905x3 ng-generic and it looks permanent there, how do I fix for s905w?

echo 45 25 1870 1055 > /sys/class/graphics/fb0/window_axis # resolve overscan emuelec screen very good
echo 0x10001 > /sys/class/graphics/fb0/free_scale # it seems not permanently

sdavides, 2022-01-07:
I create solution for s905w:
file:
/storage/.config/emuelec/scripts/setres.sh:
#!/bin/sh
fbset -fb /dev/fb0 -g 1920 1080 1920 2160 32
echo 0 0 1919 1079 > /sys/class/graphics/fb0/free_scale_axis
echo 45 25 1870 1055 > /sys/class/graphics/fb0/window_axis
chmod 444 /sys/class/graphics/fb0/window_axis
chmod 444 /sys/class/graphics/fb0/free_scale_axis
echo 0x10001 > /sys/class/graphics/fb0/free_scale # No permanent???
watch -n 1 "/storage/.config/emuelec/scripts/fix_permanent_scale.sh" &>/dev/null & # check every 1 second free_scale

/storage/.config/emuelec/scripts/fix_permanent_scale.sh:
#!/bin/bash
File=/sys/class/graphics/fb0/free_scale
if ! grep -q 0x10001 "$File"; then
  echo "diverso da 0x10001 eseguo:"
  echo "echo 0x10001 > /sys/class/graphics/fb0/free_scale" # check freescale, no permanent problem"
echo 0x10001 > /sys/class/graphics/fb0/free_scale
fi

Enjoy