EmuELEC/EmuELEC #149: T95 Max S905x3

NewAgeRipper, 2020-03-26:
Flashed a 128 gig SD card with correct image but box isn't booting from SD card. Instead it boots straight into Android. I followed ETA Primes video to the letter. Is there a work around for this?

NewAgeRipper, 2020-03-26:
Flashed a 128 gig SD card with correct image but box isn't booting from SD card. Instead it boots straight into Android. I followed ETA Primes video to the letter. Is there a work around for this?

closed 2020-03-26 by NewAgeRipper

DKAngel2000, 2020-03-26:
download terminal, the type reboot update, that shouldboot of the sdcard if you have it all setup properly

howels, 2020-03-26:
Please read the wiki (https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install) and look at existing issues https://github.com/EmuELEC/EmuELEC/issues/144 https://github.com/EmuELEC/EmuELEC/issues/141 as these are exactly what you are experiencing.

gcdoug, 2020-03-29:
There is a hidden reset switch in the AV jack. press and hold while applying power and with the SD card inserted. It should boot into EmuELEC at that point. It worked for me.