EmuELEC/EmuELEC #150: Amiga game screens are off-centre

MarkTheMorose, 2020-03-27:
I've tried about a dozen Amiga games from WHDLoad files, and all have their screens too far left and down, so I can't see what's at the far left and bottom of their screens.

I don't know how all this retro loveliness works, so I've no idea what the problem is, nor where to look (I'm new to software such as EmuELEC and Retropie).  I did try changing 'game ratio', which appears to be related to the size of the display, but no changes there made any difference.

Thanks in advance.

ramsel1508, 2020-04-16:
Observed this as well but this is not related to emuelec but rather to amiberry I guess. You can get rid of this problem if you add
 
gfx_center_horizontal=smart
gfx_center_vertical=smart

to your *.uae files

shantigilbert, 2020-06-15:
I added those lines to the configuration files, so hopefully v3.6 will fix that issue

shantigilbert, 2020-06-16:
Can you try with v3.6 ?

closed 2020-10-08 by shantigilbert