EmuELEC/EmuELEC #155: T95 Max+: Scrapped images at first, not anymore

Caninusouza, 2020-04-08:
Hi, at first I was able to get it to scrape. Now, even after starting it all over from scratch it just search for the images, but does not download them, not even for the systems it did before.  Can you please help? Not sure what I'm doing wrong. Thank you

shantigilbert, 2020-04-10:
Register on the ss website

closed 2020-04-10 by shantigilbert

Caninusouza, 2020-04-11:
SS website? What SS stands for? Thank you

chenteparao, 2020-05-13:
> Hi, at first I was able to get it to scrape. Now, even after starting it all over from scratch it just search for the images, but does not download them, not even for the systems it did before. Can you please help? Not sure what I'm doing wrong. Thank you

i can you give me the dtb for this divece please? o which one is? Thanks