EmuELEC/EmuELEC #163: Odroid N2: No Audio Out from 3.5mm jack

drpsicoloco, 2020-04-24:
I can't seem to get any audio out of the 3.5mm jack on the Odroid N2. I've seen others mention that the 3.5mm works on ubuntu. Does emuelec have support for the audio jack? Thanks for your help.

shantigilbert, 2020-05-04:
Did you find the setting on the ES menu ? can I close this issue ?

drpsicoloco, 2020-05-04:
I haven't found that option. Where is that option? Sorry, I looked around through all the Emuelec options. I must have missed it several times or don't know what I'm looking for. I found an option titles "audio device" but playing around with that didn't help. I took another look before responding and still can't find this option.

shantigilbert, 2020-05-04:
in the EmuELEC settings, set audio device to 0,1, then press B to save and reboot.

drpsicoloco, 2020-05-04:
Thank you! I didn't realize I had to reboot and wasn't sure which option worked.

closed 2020-05-04 by drpsicoloco