EmuELEC/EmuELEC #166: Bluetooth gamepad works with udev but not SDL [ppsspp]

farbirdsg, 2020-04-29:
Am using xiaomi bluetooth gamepad.
https://www.geekbuying.com/item/Original-Xiaomi-Bluetooth-Controller-Wireless-Gamepad-for-Xiaomi-Device-TV-Box-Smart-TV-Smartphone-Tablet-Black-341033.html

The gamepad works with ppsspp on ubuntu.

However in emuelec, it doesn't recognize certain buttons when configuring the mapping in ppsspp.

have already created these entries in the 
/storage/.config/ppsspp/assets/gamecontrollerdb.txt

05000000172700004431000029010000,小米蓝牙手柄,platform:Linux,a:b0,b:b1,x:b3,y:b4,back:b10,guide:b20,start:b11,leftstick:b13,rightstick:b14,leftshoulder:b6,rightshoulder:b7,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftx:a0,lefty:a1,rightx:a2,righty:a5,lefttrigger:a7,righttrigger:a6,

the entry was created using sdl2 controller mapping tool
https://retropie.org.uk/forum/topic/15577/ppsspp-controller-setup-guide-for-when-nothing-else-works

The whole episode trying to get the controller to work with ppsspp also got me experimenting using SDL2 as retroarch "joypad" driver..

Same issue happened when binding, retroarch refused to recognize some of the joypad button presses...

this problem did not occur on my ubuntu x86 setup with retroarch/ppsspp when I selected SDL for the gamepad.

do let me know which logs you'd like me to send

shantigilbert, 2020-06-15:
Does SDL2 support those characters as names?

farbirdsg, 2020-06-15:
Works in Ubuntu..(x86)

> Does SDL2 support those characters as names?

shantigilbert, 2020-06-28:
Refer to #182

closed 2020-06-28 by shantigilbert