EmuELEC/EmuELEC #167: Cannot boot past initial EMuElec screen

Bensoiree, 2020-05-03:
Hi.
I'm new to EMuElec and looking for some support, I'm trying and failing to load it on my T95z Plus running Amalogic S912. I can get it to boot and load up the initial EMuElec screen, however it fails at that point and gives the following message in the top left hand corner of the screen.
***Error in mount_storage: mount_common: could not mount LABEL=STORAGE***
###Starting debugging shell for boot step: mount_storage... Type exit to quit###

If I leave it to debug, nothing happens. 
If I type exit and quit, there is script that runs down the left hand side but lots of Failures and ultimately it stops and doesn't do anything no matter how long I wait.

Does the error about the mount storage refer to the SD card? I'm using a brand new SanDisk 64gb class 10 UHC card, I've tried older and smaller capacity cards but get the same error message. 

I've tried all of the .dtb files that work with S912 and those that will actually boot all give the same error message, so not sure what the actual issue is.

Any advice would be much appreciated, it could be something super simple that I'm just missing, I don't know. But any help or support would be great, really want to get this up and running 😊. 

Look forward to hearing from anyone, regards. 

Bensoiree 

[image]

shantigilbert, 2020-05-03:
This usually means a badly flashed SD card, try etcher, or another SD burning program.

Bensoiree, 2020-05-04:
Hi. Thanks so much for the response, I flashed the image with Win32 Disk Imager. I did find that by default it partitions the drive into two, one volume called EMuElec and the other an unformatted volume I had to delete as the box wouldn't boot if there were 2 drives present. Perhaps this has something to do with it? I'll format the card and reflash using etcher, fingers crossed it works 👍🏻😊

Another quick question, the dtb file. Would any file that gets me to the main EMuElec boot screen be fine, or does it have to be the exact dtb for my box?

Thanks again

shantigilbert, 2020-05-04:
Well... You are deleting the partition called storage, guess that is why it doesn't boot. Please read the wiki

closed 2020-05-04 by shantigilbert

ibraheeemz, 2020-11-23:
Hi,
I am facing the exact same problem shared in above snapshot. I have H96 Pro+ 3GB/32GB version and I am using the following dbt file:

H96 Pro+            3G/16G               S912         gxm_q201_3g_1gbit

while actually my H96 Pro+ box contains 32GB storage, which is not available in the provided list. Could this be the reason for not getting past this initial EMUELEC screen? 

I tried version 3.9, 3.8 and 3.7 also, all showing same error. I didn't delete any partition either. and I flash multiple times using different image burning tools.

ibraheeemz, 2020-11-23:
[image]

dm2912, 2020-11-23:
on coreelec forums people seem to either be using gxm_q201_3g.dtb or gxm_q200_3g.dtb but theres a few people have had issues. trouble with these unbranded generic boxes like X96 Pro, Max etc is they very often have completely different parts in them from batch to batch.

ibraheeemz, 2020-11-23:
> on coreelec forums people seem to either be using gxm_q201_3g.dtb or gxm_q200_3g.dtb but theres a few people have had issues. trouble with these unbranded generic boxes like X96 Pro, Max etc is they very often have completely different parts in them from batch to batch.

so basically, the error is related to the dtb file?

AlienatedDiego, 2020-11-23:
It is said that these boxes sometimes change components depending on the batch or where they are manufactured, sometimes even with a different soc or fake ram. Try all the q200 / q201 .dtb in the device trees folder until you find a valid one. On the other hand There are user reports that this device works with gxm_q201_3g_1gbit