EmuELEC/EmuELEC #168: How to force 720p

ArturNadolski, 2020-05-04:
I've NEC 1980 FXi 19" 5:4 LCD monitor. I would like to set it to 720p mode as native for EmuELEC.
This monitor doesn't support 1080p so after fresh EmuELEC start I'm getting image like at attached pictures.
[image]
[image]

ArturNadolski, 2020-05-04:
How to add splash and intro for 720p? Where to put files and how to name it to system recognize and use it instead of original ones in 1080p?

shantigilbert, 2020-05-04:
Read the wiki on how to change the resolution. 

There is no support for changing intro video or splash.

closed 2020-05-04 by shantigilbert