EmuELEC/EmuELEC #171: Switching to external sound card not working on 3.5 Odroid N2

qbertaddict, 2020-05-06:
On 3.4 I am able to switch audio to device 1,1

On 3.5 I select the same device it does not switch over. 

I did a aplay -l and the device number did not change between the versions

This is on the Odroid N2 4 gig version

shantigilbert, 2020-05-07:
does the USB sound card get loaded correctly? 

what is the result of

`dmesg | grep usb`
`lsusb`

qbertaddict, 2020-05-08:
EmuELEC:~ # dmesg | grep usb
[    0.472880@2] usbcore: registered new interface driver usbfs
[    0.472912@2] usbcore: registered new interface driver hub
[    0.472972@2] usbcore: registered new device driver usb
[    0.552444@4] usbcore: registered new interface driver cdc_acm
[    0.552482@4] usbcore: registered new interface driver usb-storage
[    0.552541@4] usbcore: registered new interface driver usbserial
[    0.552567@4] usbcore: registered new interface driver usbserial_generic
[    0.552590@4] usbserial: USB Serial support registered for generic
[    0.554310@2] usbcore: registered new interface driver usbhid
[    0.554311@2] usbhid: USB HID core driver
[    0.652696@2] amlogic-new-usb2-v2 ffe09000.usb2phy: USB2 phy probe:phy_mem:0xffe09000, iomap phy_base:0xffffff80083fa000
[    0.652838@2] amlogic-new-usb3-v2 ffe09080.usb3phy: USB3 phy probe:phy_mem:0xffe09080, iomap phy_base:0xffffff80083fe080
[    1.053863@0] usbcore: registered new interface driver snd-usb-audio
[    1.321873@2] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.321896@2] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.321898@2] usb usb1: Product: xHCI Host Controller
[    1.321900@2] usb usb1: Manufacturer: Linux 4.9.113 xhci-hcd
[    1.321901@2] usb usb1: SerialNumber: xhci-hcd.0.auto
[    1.322291@2] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.322324@2] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[    1.322326@2] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.322328@2] usb usb2: Product: xHCI Host Controller
[    1.322329@2] usb usb2: Manufacturer: Linux 4.9.113 xhci-hcd
[    1.322331@2] usb usb2: SerialNumber: xhci-hcd.0.auto
[    1.333092@2] dwc_otg: usb0: type: 2 speed: 0, config: 0, dma: 0, id: 0, phy: ffe09000, ctrl: 0
[    1.593908@2] amlogic-new-usb2-v2 ffe09000.usb2phy: ---Set port(0) tuning for host cf(xhci_hub_control)--
[    1.653888@2] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[    1.796239@2] usb 1-1: New USB device found, idVendor=05e3, idProduct=0610
[    1.796242@2] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.796244@2] usb 1-1: Product: USB2.1 Hub
[    1.796245@2] usb 1-1: Manufacturer: GenesysLogic
[    1.914015@0] usb 2-1: new SuperSpeed USB device number 2 using xhci-hcd
[    1.936580@0] usb 2-1: New USB device found, idVendor=05e3, idProduct=0620
[    1.936586@0] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.936589@0] usb 2-1: Product: USB3.1 Hub
[    1.936591@0] usb 2-1: Manufacturer: GenesysLogic
[    2.149896@2] usb 1-1.2: new high-speed USB device number 3 using xhci-hcd
[    2.275262@2] usb 1-1.2: New USB device found, idVendor=152d, idProduct=2336
[    2.275265@2] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=5
[    2.275267@2] usb 1-1.2: Product: JM20336 SATA, USB Combo
[    2.275269@2] usb 1-1.2: Manufacturer: JMicron
[    2.275271@2] usb 1-1.2: SerialNumber: 2D1B10457888
[    2.276561@2] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[    2.276774@2] scsi host0: usb-storage 1-1.2:1.0
[    2.377887@2] usb 1-1.3: new full-speed USB device number 4 using xhci-hcd
[    2.503035@2] usb 1-1.3: New USB device found, idVendor=045e, idProduct=02dd
[    2.503037@2] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.503039@2] usb 1-1.3: Product: Controller
[    2.503041@2] usb 1-1.3: Manufacturer: Microsoft
[    2.503042@2] usb 1-1.3: SerialNumber: 7EED894B3A80
[    2.506480@2] usb 1-1.3: Unsupported device
[    2.506561@2] usb 1-1.3: Unsupported device
[    2.506599@2] usb 1-1.3: Unsupported device
[   11.738123@3] input: Microsoft X-Box One pad (Firmware 2015) as /devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.3/1-1.3:1.0/input/input3
[   11.738282@3] usbcore: registered new interface driver xpad
[   37.921915@0] usb 1-1.4: new full-speed USB device number 5 using xhci-hcd
[   38.047500@0] usb 1-1.4: New USB device found, idVendor=0d8c, idProduct=013c
[   38.047507@0] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   38.047509@0] usb 1-1.4: Product: USB PnP Sound Device
[   38.047512@0] usb 1-1.4: Manufacturer: C-Media Electronics Inc.
[   38.195058@0] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.3/0003:0D8C:013C.0001/input/input4
[   38.254633@0] hid-generic 0003:0D8C:013C.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-xhci-hcd.0.auto-1.4/input3
EmuELEC:~ # lsusb
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0d8c:013c C-Media Electronics, Inc. CM108 Audio Controller
Bus 001 Device 004: ID 045e:02dd Microsoft Corp. Xbox One Controller (Firmware 2015)
Bus 001 Device 003: ID 152d:2336 JMicron Technology Corp. / JMicron USA Technology Corp. Hard Disk Drive
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

shantigilbert, 2020-11-26:
Is this still an issue?

closed 2021-04-24 by shantigilbert