EmuELEC/EmuELEC #172: Not all ROMS recognised

Bensoiree, 2020-05-06:
I managed to get EMUElec running on my T95z Plus, thanks for the help with the booting issues Shantigilbert.

I'm now finding issues with the ROMS I'm putting on the SD card, they're listed in the wiki as being the correct file type (N64, z64 etc) and are in the correct folder in the ROMS section of the storage, but not all of the ROMS are being listed in the available games when I boot up. In fact very few are, and I've loaded hundreds on there. I tried refreshing the games list but thag hasn't worked, its as if they aren't there at all. 

Is there any known reason for this? Will it support multiple file types in the same folder or do they all have to be the same(all N64 rather than some N64 and some z64 for example) ? Have I loaded the wrong ROMS for my box (S912) ? Is there such a thing as an incorrect ROM? 

Any help would be much appreciated, loving what I've seen of emuelec so far. Keep up the good work 👍🏻

shantigilbert, 2020-05-06:
The only reason is there are in the incorrect path, check your paths and they are case sensitive. if they are not listed then the rom is not setup on es_ systems.cfg

Bensoiree, 2020-05-07:
Thanks for the reply, as far as I can see they have been put in the correct folders? I can only see one folder for n64 for example, so this is where I have put all the ROMS for that console. Before I loaded a huge library of ROMS on there I added just one or two for each console to see if they worked, they came from a different source and were added to the path folder as a zip/rar file. They work and are available to play, however the rest that are not in zip/rar files are not recognised? Wasn't sure if that would be a problem? Can I update the es_systems.cfg?

Thanks, and sorry for all the questions but this is how I learn :-)

chenteparao, 2020-05-07:
Hi, I have a similar issue, My system (3.5 on T95 S1) recognized all roms of N64, but non works with MUPEN64PLUS NEXT, just work with MUPEN64PLUS and PARALLEL but it seems very bad, any comment? thanks

shantigilbert, 2020-05-07:
> Hi, I have a similar issue, My system (3.5 on T95 S1) recognized all roms of N64, but non works with MUPEN64PLUS NEXT, just work with MUPEN64PLUS and PARALLEL but it seems very bad, any comment? thanks

that is certainly not the same issue

shantigilbert, 2020-05-07:
> Thanks for the reply, as far as I can see they have been put in the correct folders? I can only see one folder for n64 for example, so this is where I have put all the ROMS for that console. Before [...]

Have you read the wiki? it has a;ll the supported extensions listed for all platforms. Like I said if they do not show up then your extensions are either bad or they are not in the correct folder path.

Bensoiree, 2020-05-07:
Thanks for the reply. Like I said in my original post I have read the wiki, and everything I have dropped onto the memory card is in the correct folder path and is a supported file type. But, they're not listed when I boot up. I'm sorry if these questions seem stupid, but I have read the wiki, and I have looked through all the open and closed issues for something like this being asked and answered but can't see anything, which is why I posted.

I'm new to this and don't have the knowledge you do, when you say the extensions are bad do you mean the file type is not correct? Or the files are corrupt? 

Sorry if these are really stupid questions to you, but I don't have the answers and can't seem to find them on here or on Google. 

Thanks in advance

shantigilbert, 2020-05-07:
```
<system>
		<name>n64</name>
		<fullname>Nintendo 64</fullname>
		<path>/storage/roms/n64</path>
		<extension>.z64 .Z64 .n64 .N64 .v64 .V64 .zip .ZIP .7z .7Z</extension>
		<command>/emuelec/scripts/emuelecRunEmu.sh %ROM% -P%SYSTEM% --controllers="%CONTROLLERSCONFIG%"</command>
		<platform>n64</platform>
		<theme>n64</theme>
	</system>
```
This is whats listed on n64 as extensiones, if your extensions are NOT there, then it will not show up. there is not much else to it.

chenteparao, 2020-05-07:
> > Hi, I have a similar issue, My system (3.5 on T95 S1) recognized all roms of N64, but non works with MUPEN64PLUS NEXT, just work with MUPEN64PLUS and PARALLEL but it seems very bad, any comment? thanks
> 
> that is certainly not the same issue

Do you wan that i open a new issue? certainly put .zip .n64 and .z64 but nothing with Mupen64plus next

shantigilbert, 2020-05-07:
> Do you wan that i open a new issue? certainly put .zip .n64 and .z64 but nothing with Mupen64plus next

Yeah his issue is that the ROMS are not appearing, yours is just performance, which is probably normal depending on your device, so no need to open another issue.

sauceisgood, 2020-05-07:
I solve this by change the emulator... And for now, it works smooth... I have an x96 mini 2g

Bensoiree, 2020-05-07:
OK, so here's an update. 

Sort version, I got it working. 

Long version. Whilst trying and failing to get the ROMS to be recognised, the SD card somehow got corrupted. Emuelec didn't load and I get the black screen saying file corrupted. Pulled the SD and formatted it, various issues later I managed to get it to mount with one ntfs volume, I used a combination of Lunix File Systems for Windows and Disk Manager to fix the corrupted sections and then tried to flash the img file again. For some reason etcher wouldn't do it saying the drive was corrupted still, but Win32 Disk Imager didn't find the same issues and flashed the img successfully (I previously couldn't get Win32 to work which is why I tried etcher). After booting successfully for the first time after the issues, I went to load the ROMS into the correct paths on the SD and once I booted it back up again BOOM! There they were! Scraped the images and off I went. Still not sure why they loaded this time and not before, they're the same ROMS loaded the same way, to and from the same place.... But that doesn't matter now. I now have 301 ROMS to play with.

Thanks for all the help and sorry for the dumb questions

closed 2020-05-07 by Bensoiree

shantigilbert, 2020-05-08:
No problem! glad you got it fixed!