EmuELEC/EmuELEC #173: Unable to load roms off external USB drive

joesnarky, 2020-05-07:
following the steps here
https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-USB-media

Emuelec is seeing my USB
I can map the drive and see the data no problem. Lots of roms
\\192.168.1.114\USB Media\USB\roms\dreamcast
\\192.168.1.114\USB Media\USB\roms\megadrive
\\192.168.1.114\USB Media\USB\roms\mame

I copied the roms from an Emulec build I could not get booting on my T95 so I am pretty sure all the system folder names are good.
I did create an empty file called "emulecroms" in the roms folder. 

I am using the latest version of EmuElec 3.5
My box is the T95 Max+  s905x3 4gb 
Im using a 128GB USB stick formated as NTFS

Do I need to label the USB stick with a specific name or something or remove the ROMS folder on the SD card since it says specifically you cant run roms from both SD and USB?

shantigilbert, 2020-05-07:
> "emulecroms

its called `emuelecroms` and make sure it does not have a hidden extension and its all lowercase

joesnarky, 2020-05-08:
omg I can't believe I did that. Tried to be extra careful and cut and paste. Working perfectly. Live very sweet. plugged in the usb stick and emultionstation reloaded on its own with the new roms. And seems all the media and images load super fast.   Thanks.

closed 2020-05-08 by joesnarky