EmuELEC/EmuELEC #174: 8bitdo M30 bluetooth controller doesn't pair after restarting the device

blinciki-lol, 2020-05-07:
8bitdo M30 bluetooth controller doesn't pair after restarting EmuElec or after the controller automatically goes to sleep

One note, I couldn't pair the controller in EmuElec's UI, so I followed the bluetooth guide, which helped to pair the controller: https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth 

Steps to reproduce the behavior:
1. Following the "Bluetooth" guide, successfully pair the 8bitdo M30 controller with the device.
2. Everything works as expected, till the device is restarted
2.1 Or, in case another wireless controller is being used, this one goes to sleep and no longer pairs after that.
3. Either restart or shutdown the device then turn it on.
4. Try to turn on the controller and it no longer pairs with the device.
5. To solve the problem, all bluetooth controllers need to be "forgotten" and follow again the bluetooth guide.

Expected behavior
After a restart or using other wireless controllers, the controller should still pair with the device.

 - EmuELEC Version 3.5

Device:
 - SOC: S922X 
 - Brand: Odroid
 - Model: N2
 - Version: ?

Additional context
I tried both EmuELEC version 3.4 and 3.5 and the behavior persists. Made tests also with dualshock 4 (V1 and V2) both pair directly in EmuELEC, work without any issues and pair after restart or after adding more controllers.

I've read about this behavior and noticed that it's persistent for the 8bitdo controllers (on other retro distros too). On the other hand, it's connects without any issues to PC or an android phone/tablet.

I've found only this solution, that RetroPie has for this this type of controllers: https://github.com/RetroPie/RetroPie-Setup/wiki/8Bitdo-Controller
In the following guide, they present a tutorial on how to set up these controllers.
From the first step till the 11th step - it follow the "same" steps as the "Bluetooth" guide, it's just done in the RetroPie and not by SSH. I think the next steps are the ones that solve this problem - like setting udev rule  in order for Emulation Station to "see" the controller when you restart your Raspberry Pi and changing the connect mode from default to boot or background. 

I don't know how to make these changes manually and save them. Could you please provide a guide on how to do them or some advice on how to do it?

shantigilbert, 2020-05-07:
There seems to be an unknown issue on certain boxes (BT chips) that this seems to happen, unfortunately I haven't found a way to resolve this, it works for some and it does not for others. 

EmuELEC automatically searches for BT pads on boot (or it should), so that is not the issue.

farbirdsg, 2020-05-08:
I usually pair them this manner in bluetoothctl

pair <BTMAC>
trust <BTMAC>
connect <BTMAC>

shantigilbert, 2020-07-13:
Can you try latest 3.7  test version? also use the Bluetooth menu in controller settings

blinciki-lol, 2020-07-13:
I don't have net now, but should have back it soon. I'll let you know once I've tested the new release.

blinciki-lol, 2020-07-15:
I've done multiple tests and shortly - everything works. There's only one observed problem, after connecting wirelessly dualshock 4 v1, it's no longer possible to connect the 8bitdo controller (removing the dualshock saved pairing solves the problem).

The following setups work:
Wired controller + 8bitdo wireless
Wired controller + 8bitdo wireless + dualshock 4 v2 (or multiple v2s)
8bitdo wireless + dualshock 4 v2 (or multiple v2s)
8bitdo wireless (only one controller)
*dualshock 4 v1 + dualshock 4 v2 (or multiple v2s)

This setup doesn't work:
dualshock 4 v1 + 8bitdo
After successfully pairing and connecting the first generation of dualshock 4, the 8bitdo controller is no longer seen in EmuElec (neither in the Settings menu nor in the new script )

Regarding, the automatic connection it's worked from the first try in the Settings menu (Note: only the D-input mode works). In the script section it also worked, and both D-input and X-input were to pair. Thank you for these options! Now technically you don't need a pc to setup these controllers.

Notes:
1) Starting the controller on startup connects the controller faster, though the controller still connected, even if I turned it on later, but it took a little longer to pair in comparison when you turn it on - on startup.
2) I only had two cases when the connection dropped, the first one was a hard reset, because retroarch froze - solved with script (removed the pairing and connected again) and the second case was on the third restart of the system(in the beginning when I've done these test) - solved again in the settings menu.
After multiple, resets, shutdown, restarts the connection didn't drop. Fortunately, I couldn't recreate again this behavior.
3) It's better to avoid using the dualshock 4 version 1, as the system no longer recognizes the 8bitdo controller. This is strange, but it doesn't matter, as the second version of the dualshock doesn't have this problem.

Thank you for all your work!! :)

closed 2020-07-16 by blinciki-lol