EmuELEC/EmuELEC #177: Change DOSBox to DOSBox-x

andrewg78, 2020-05-13:
**Is your feature request related to a problem? Please describe.**
Yes. In the original DosBox there is no feature to save and load a game state 

**Describe the solution you'd like**
Instead of the original dosbox version, compile a fork called dosbox-x 
https://github.com/joncampbell123/dosbox-x

shantigilbert, 2020-05-13:
I tried Dosbos-x, unfortunately I had problems with how the games are displayed (window way to small) until I figure it out, I have no choice but to keep using dosbox :(

andrewg78, 2020-05-13:
I would like to play a bit with this project.  How do u execute a cross compilation for Amlogic generic ?
Could you provide some commands here?

shantigilbert, 2020-05-14:
Everything you need to compile is on the readme including dependencies and commands.

shantigilbert, 2020-06-28:
Dosbox-x has been added

closed 2020-06-28 by shantigilbert