EmuELEC/EmuELEC #179: DOSBox MT32 emulation crashes

mbarrio, 2020-05-16:
When selecting mididevice=mt32 (properly configured with mt32.romdir pointing to the ROMs), DOSBox crashes with the error:

dosbox: symbol lookup error: /usr/lib/libmt32emu.so.2: undefined symbol: _Z15log10f_neon_sfpf

**To Reproduce**
Steps to reproduce the behavior:
1. Edit the game's .conf file and choose mididevice=mt32 and mt32.romdir.
2. Launch DOSBox and use the game's setup executable to configure it to use MT32 music sound card.
3. Launch DOSBox and run the game.
4. See the error.

**Expected behavior**
DOSBox should not crash when using MT32 MIDI music option.

**DOSBox output**
DOSBox version SDL2
Copyright 2002-2019 DOSBox Team, published under GNU GPL.

CONFIG: Loading primary settings from config file dosbox-SDL2.conf
SDL: Current window pixel format: SDL_PIXELFORMAT_RGBX8888
MIXER: Got different values from SDL: freq 44100, blocksize 512
MIDI: Opened device:mt32
Using joystick Logitech Gamepad F710 with 6 axes, 11 buttons and 1 hat(s)
Using driver "opengles2" for renderer
dosbox: symbol lookup error: /usr/lib/libmt32emu.so.2: undefined symbol: _Z15log10f_neon_sfpf

**EmuELEC version:**
 - 3.5

**Device (please complete the following information):**
 - SOC: S095x3
 - Brand: Ninkbox
 - Model: N8 Plus

shantigilbert, 2020-05-16:
Thank you for the report. I will look into it, this might be a munt issue, since I remember a while back I ran into something similar.

shantigilbert, 2020-06-16:
Can you try with v3.6?

mbarrio, 2020-06-16:
I have tested v3.6 with Risky Woods game and it is working now. Thank you very much!

closed 2020-06-16 by mbarrio