EmuELEC/EmuELEC #180: PPSSPPA [OGA] Analog stick sensitivity issues

zombiekutez, 2020-05-16:
I'm using Odroid Go Advance, running Emuelec 3.5, update May.
I running PPSSPPA core, when moving the stick at max distance, ppsspp is acting as if the stick is not at its maximum distance. This makes movement in games like the 3rd person game very awkward as you are forced to walk instead of run in certain directions, the character all way walking gameplay, but some time they run if moving the stick max to left and right, up down still alway walking, many 3rd person game have that issues, Prince of Persia, Splinter Cell, Crash Bandicoot. Other os like The Retro Arena have that issues too. I try a lot of things, like set the sensetivity in sdl, sensetivity in Lakka (this really fix that issues if we use ppsspp_libretro, but that core had more glitch and bad performance) it still happen, but for now PPSSPPSDL2 is already port to OGA (https://forum.odroid.com/viewtopic.php?f=194&t=38045), i have used that on odroid gamestation tubro, PPSSPPSDL2 get much better performance, fix analog sensetivity issues, and many things better. Can you update this core on next release, or somethings to fix that issues, please.

shantigilbert, 2020-05-16:
This should be asked in the main PPSSPP-go2 repo as it has the controls hard-coded
https://github.com/OtherCrashOverride/ppsspp-go2

The link you provided is for SDL2 and not PPSSPP, that SDL2 is already included in EmuELEC

But if you are asking to include the upstream PPSSPP I guess I could try it when I get some free time. 

Otherwise can you elaborate on what version you said you tried?

zombiekutez, 2020-05-17:
> This should be asked in the main PPSSPP-go2 repo as it has the controls hard-coded
> https://github.com/OtherCrashOverride/ppsspp-go2
>
> The link you provided is for SDL2 and not PPSSPP, that SDL2 is [...]

[image]
Okay it exactly is v1.9.3-481-gd0e2aa3a4 ver. I found this file from the image os.
[image]
You can view they file in here. https://drive.google.com/file/d/1valrtDT7kHKAD9qcxNp0vMGHsW6U1Zz9/view?usp=drivesdk
I hope this can help you. This image base from debian buster. https://forum.odroid.com/viewtopic.php?f=193&t=37399.
Thank.

shantigilbert, 2020-06-10:
PPSSPPSDL is going to be used instead of PPSSPP-GO for the next release

shantigilbert, 2020-07-13:
https://github.com/OtherCrashOverride/ppsspp-go2/issues/2 there is an issue already oppened in PPSSPP-GO2 repo, hopefully it gets fixed there.

closed 2020-10-08 by shantigilbert