EmuELEC/EmuELEC #182: emulationstation hotkey not working in games.. which cfg file to check?

farbirdsg, 2020-05-22:
**Describe the bug**
am not able to get emulationstation hotkeys working in games

**To Reproduce**
start game, press hotkey + start

quit game and goes back emulationstation ui

**Screenshots**
not applicable

**EmuELEC version:**
3.5

**Device (please complete the following information):**
Beelink GT King Pro
S922

**Additional context**
what cfg files do I have to check?
am ok with ssh into the system [ linux user here ].

i have already set the hotkey button when configuring the joypad in the emulationstation joypad config.
tried more than one type of joypad

shantigilbert, 2020-05-22:
I am not sure what the problem is... hotkey + start means quit game and go back to ES... what where you expecting ?

farbirdsg, 2020-05-22:
> I am not sure what the problem is... hotkey + start means quit game and go back to ES... what where you expecting ?

I am not able to quit back to ES

coach1988, 2020-05-24:
Afaik EmuELEC copies over the initial first gamepad config to RA.i think you are looking for the binding of the hotkey in retroarch.

shantigilbert, 2020-05-25:
First of all, which emulator? second, emuelec.log would be very helpful here.

thiagoteles, 2020-06-01:
Using any emulator, in any game I cant leave the games too. I`ve reconfigured the controller to scratch, defined the hotkey to select button and when I press start+select (two times) nothing happens.

es_log: 

`2020-06-01 01:38:43	INFO	Configuring device 0 (Microntek              USB Joystick          ).
2020-06-01 01:38:46	INFO	  Mapping [Hat 0 up] -> Up
2020-06-01 01:38:48	INFO	  Mapping [Hat 0 down] -> Down
2020-06-01 01:38:48	INFO	  Mapping [Hat 0 left] -> Left
2020-06-01 01:38:49	INFO	  Mapping [Hat 0 right] -> Right
2020-06-01 01:38:49	INFO	  Mapping [Button 9] -> Start
2020-06-01 01:38:50	INFO	  Mapping [Button 8] -> Select
2020-06-01 01:38:51	INFO	  Mapping [Button 1] -> A
2020-06-01 01:38:52	INFO	  Mapping [Button 2] -> B
2020-06-01 01:38:53	INFO	  Mapping [Button 0] -> X
2020-06-01 01:38:53	INFO	  Mapping [Button 3] -> Y
2020-06-01 01:38:54	INFO	  Mapping [Button 4] -> LeftShoulder
2020-06-01 01:38:55	INFO	  Mapping [Button 5] -> RightShoulder
2020-06-01 01:38:56	INFO	  Mapping [Button 6] -> LeftTrigger
2020-06-01 01:38:56	INFO	  Mapping [Button 7] -> RightTrigger
2020-06-01 01:39:06	INFO	  Mapping [Axis 1-] -> LeftAnalogUp
2020-06-01 01:39:10	INFO	  Mapping [Axis 1+] -> LeftAnalogDown
2020-06-01 01:39:16	INFO	  Mapping [Axis 0-] -> LeftAnalogLeft
2020-06-01 01:39:17	INFO	  Mapping [Axis 0+] -> LeftAnalogRight
2020-06-01 01:39:20	INFO	  Mapping [Axis 3-] -> RightAnalogUp
2020-06-01 01:39:23	INFO	  Mapping [Axis 3+] -> RightAnalogDown
2020-06-01 01:39:25	INFO	  Mapping [Axis 2-] -> RightAnalogLeft
2020-06-01 01:39:26	INFO	  Mapping [Axis 2+] -> RightAnalogRight
2020-06-01 01:39:29	INFO	  Mapping [Button 11] -> HotKeyEnable
2020-06-01 01:39:35	INFO		bash /storage/.emulationstation/scripts/inputconfiguration.sh
2020-06-01 01:39:37	INFO	1 INPUT P1NAME INPUT P1GUID
2020-06-01 01:39:37	WARNING	adding entry for player0 (not selected): Microntek              USB Joystick            03000000790000000600000010010000
2020-06-01 01:39:37	WARNING	adding default entry for player 0(selected : true)
2020-06-01 01:39:37	INFO	2 INPUT P2NAME INPUT P2GUID
2020-06-01 01:39:37	WARNING	adding entry for player1 (not selected): Microntek              USB Joystick            03000000790000000600000010010000
2020-06-01 01:39:37	WARNING	adding default entry for player 1(selected : true)
2020-06-01 01:39:37	INFO	3 INPUT P3NAME INPUT P3GUID
2020-06-01 01:39:37	WARNING	adding entry for player2 (not selected): Microntek              USB Joystick            03000000790000000600000010010000
2020-06-01 01:39:37	WARNING	adding default entry for player 2(selected : true)
2020-06-01 01:39:37	INFO	4 INPUT P4NAME INPUT P4GUID
2020-06-01 01:39:37	WARNING	adding entry for player3 (not selected): Microntek              USB Joystick            03000000790000000600000010010000
2020-06-01 01:39:37	WARNING	adding default entry for player 3(selected : true)
2020-06-01 01:39:37	INFO	5 INPUT P5NAME INPUT P5GUID
2020-06-01 01:39:37	WARNING	adding entry for player4 (not selected): Microntek              USB Joystick            03000000790000000600000010010000
2020-06-01 01:39:37	WARNING	adding default entry for player 4(selected : true)
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile:selected] - expected "ninepatch", got "text"
2020-06-01 01:39:49	WARNING	 requested mismatched theme type for [Tiles.gridtile] - expected "video", got "text"
[... cut ...]
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Written to playlist file: /storage/.config/retroarch/content_history.lpl
[INFO] [Environ]: SET_GEOMETRY.
[INFO] [Environ]: SET_GEOMETRY.
[INFO] [Environ]: SET_GEOMETRY.
`

shantigilbert, 2020-06-16:
post your `/tmp/joypad/nameofyourgamepad.cfg` and `/storage/.emulationstation/es_input.cfg`

farbirdsg, 2020-06-16:
/storage/.config/emulationstation/es_input.cfg

<?xml version="1.0"?>
<inputList>
  <inputAction type="onfinish">
    <command>bash /storage/.emulationstation/scripts/inputconfiguration.sh</command>
  </inputAction>
  <inputConfig type="joystick" deviceName="Sony PLAYSTATION(R)3 Controller">
    <input name="a" type="button" id="13" value="1"/>
    <input name="b" type="button" id="14" value="1"/>
    <input name="down" type="button" id="6" value="1"/>
    <input name="hotkeyenable" type="button" id="16" value="1"/>
    <input name="left" type="button" id="7" value="1"/>
    <input name="leftanalogdown" type="axis" id="1" value="1"/>
    <input name="leftanalogleft" type="axis" id="0" value="-1"/>
    <input name="leftanalogright" type="axis" id="0" value="1"/>
    <input name="leftanalogup" type="axis" id="1" value="-1"/>
    <input name="leftshoulder" type="button" id="10" value="1"/>
    <input name="leftthumb" type="button" id="1" value="1"/>
    <input name="lefttrigger" type="button" id="8" value="1"/>
    <input name="right" type="button" id="5" value="1"/>
    <input name="rightanalogdown" type="axis" id="3" value="1"/>
    <input name="rightanalogleft" type="axis" id="2" value="-1"/>
    <input name="rightanalogright" type="axis" id="2" value="1"/>
    <input name="rightanalogup" type="axis" id="3" value="-1"/>
    <input name="rightshoulder" type="button" id="11" value="1"/>
    <input name="rightthumb" type="button" id="2" value="1"/>
    <input name="righttrigger" type="button" id="9" value="1"/>
    <input name="select" type="button" id="0" value="1"/>
    <input name="start" type="button" id="3" value="1"/>
    <input name="up" type="button" id="4" value="1"/>
    <input name="x" type="button" id="12" value="1"/>
    <input name="y" type="button" id="15" value="1"/>
  </inputConfig>
  <inputConfig type="joystick" deviceName="GZ RX CO.,LTD. RX Gamepad">
    <input name="start" type="button" id="9" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="a" type="button" id="4" value="1"/>
    <input name="b" type="button" id="0" value="1"/>
    <input name="hotkeyenable" type="button" id="11" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="x" type="button" id="1" value="1"/>
    <input name="select" type="button" id="10" value="1"/>
    <input name="y" type="button" id="2" value="1"/>
    <input name="left" type="hat" id="0" value="8"/>
    <input name="rightanalogright" type="button" id="9" value="1"/>
    <input name="lefttrigger" type="button" id="3" value="1"/>
    <input name="righttrigger" type="button" id="6" value="1"/>
  </inputConfig>
  <inputConfig type="joystick" deviceName="Sony Interactive Entertainment Wireless Controller" deviceGUID="030000004c050000cc09000011010000">
    <input name="rightanalogdown" type="axis" id="5" value="1"/>
    <input name="leftshoulder" type="button" id="4" value="1"/>
    <input name="start" type="button" id="9" value="1"/>
    <input name="leftanalogup" type="axis" id="1" value="-1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="a" type="button" id="2" value="1"/>
    <input name="b" type="button" id="1" value="1"/>
    <input name="hotkeyenable" type="button" id="12" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="leftanalogright" type="axis" id="0" value="1"/>
    <input name="leftanalogdown" type="axis" id="1" value="1"/>
    <input name="rightshoulder" type="button" id="5" value="1"/>
    <input name="righttrigger" type="button" id="7" value="1"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="x" type="button" id="3" value="1"/>
    <input name="select" type="button" id="8" value="1"/>
    <input name="leftanalogleft" type="axis" id="0" value="-1"/>
    <input name="y" type="button" id="0" value="1"/>
    <input name="left" type="hat" id="0" value="8"/>
[... 264 lines cut ...]
    <input name="start" type="button" id="16" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="x" type="button" id="10" value="1"/>
    <input name="y" type="button" id="9" value="1"/>
  </inputConfig>
  <inputConfig type="joystick" deviceName="&#x5C0F;&#x7C73;&#x84DD;&#x7259;&#x624B;&#x67C4;" deviceGUID="05000000172700004431000029010000">
    <input name="a" type="button" id="7" value="1"/>
    <input name="b" type="button" id="6" value="1"/>
    <input name="down" type="hat" id="0" value="4"/>
    <input name="hotkeyenable" type="button" id="12" value="1"/>
    <input name="left" type="hat" id="0" value="8"/>
    <input name="leftshoulder" type="button" id="12" value="1"/>
    <input name="leftthumb" type="button" id="19" value="1"/>
    <input name="lefttrigger" type="button" id="14" value="1"/>
    <input name="right" type="hat" id="0" value="2"/>
    <input name="rightshoulder" type="button" id="13" value="1"/>
    <input name="rightthumb" type="button" id="20" value="1"/>
    <input name="righttrigger" type="button" id="15" value="1"/>
    <input name="select" type="button" id="17" value="1"/>
    <input name="start" type="button" id="16" value="1"/>
    <input name="up" type="hat" id="0" value="1"/>
    <input name="x" type="button" id="10" value="1"/>
    <input name="y" type="button" id="9" value="1"/>
  </inputConfig>
</inputList>

farbirdsg, 2020-06-16:
/storage/joypads/小米蓝牙手柄.cfg

`input_device = "小米蓝牙手柄"
input_driver = "udev"
input_vendor_id = "10007"
input_product_id = "12612"
input_l3_btn = "19"
input_l_btn = "12"
input_load_state_btn = "12"
input_start_btn = "16"
input_exit_emulator_btn = "16"
input_up_btn = "h0up"
input_volume_up_btn = "h0up"
input_a_btn = "7"
input_b_btn = "6"
input_reset_btn = "6"
input_enable_hotkey_btn = "12"
input_down_btn = "h0down"
input_volume_down_btn = "h0down"
input_r_btn = "13"
input_save_state_btn = "13"
input_r2_btn = "15"
input_hold_fast_forward_btn = "15"
input_r3_btn = "20"
input_fps_toggle_btn = "20"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_x_btn = "10"
input_menu_toggle_btn = "10"
input_select_btn = "17"
input_y_btn = "9"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_l2_btn = "14"
input_rewind_btn = "14"`

shantigilbert, 2020-06-16:
Unfortunately I do not know how to handle CN characters in the auto configurations. 

`&#x5C0F;&#x7C73;&#x84DD;&#x7259;&#x624B;&#x67C4;` and `小米蓝牙手柄` so someone else will need to figure this one out. Sorry

farbirdsg, 2020-06-16:
> Unfortunately I do not know how to handle CN characters in the auto configurations.
> 
> `&#x5C0F;&#x7C73;&#x84DD;&#x7259;&#x624B;&#x67C4;` and `小米蓝牙手柄` so someone else will need to figure this one out. Sorry
> 
> Also your es_input.cfg is wrong.

can elaborate the part abt es_input.cfg being wrong?

shantigilbert, 2020-06-16:
> > Unfortunately I do not know how to handle CN characters in the auto configurations.
> > `&#x5C0F;&#x7C73;&#x84DD;&#x7259;&#x624B;&#x67C4;` and `小米蓝牙手柄` so someone else will need to figure this one out. Sorry
> > Also your es_input.cfg is wrong.
> 
> can elaborate the part abt es_input.cfg being wrong?

unless it was a paste error: 

```
 </inputConfig>
  <input name="a" type="button" id="7" value="1"/>
```

there is a place in the file that is not correctly formatted

farbirdsg, 2020-06-17:
in my es_temporaryinput.cfg, this line was present, v similar to the line shown in es_inputs.cfg but the unicode characters were correctly translated

` <inputConfig type="joystick" deviceName="小米蓝牙手柄" deviceGUID="0500>
                <input name="a" type="button" id="7" value="1" />`

am curious why the characters did not get ported over to the es_inputs.cfg

`<inputConfig type="joystick" deviceName="&#x5C0F;&#x7C73;&#x84DD;&#x7259;&#x624B;&#x67C4;" deviceGUID="05000000172700004431000029010000">`

shantigilbert, 2020-06-24:
The thing is, I use sed to detect controllers, since the CN characters are unfortunately not standard, I do not know how to handle that with sed, so the only solution I can offer at the moment is to use another gamepad, one that does not have CN characters in in it's internal name. If I find a solution I will fix it, but I will be honest, its not a priority for me. Or if someone else can help with this then I would appreciate a PR

TheLastProject, 2020-07-22:
@shantigilbert Could you link to the part of the code that does this? This seems like a fun little challenge for me to take a look at

shantigilbert, 2020-07-22:
> 
> 
> @shantigilbert Could you link to the part of the code that does this? This seems like a fun little challenge for me to take a look at

ES creates a temp configuration which in turn gets handled by these scripts

https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec-emulationstation/config/scripts/configscripts

it seems the issue is on either the emulationstation.sh or the retroarch.sh file.

TheLastProject, 2020-07-22:
I've been staring at this for an hour and I can't see any place where the DEVICE_NAME read with xmlstarlet gets changed at all :(

Langerz82, 2021-09-06:
Hey, sorry for Necro but I think the dev branch should have fixed this issue. The problem was gptokeyb was not loading the gamecontrollerdb.txt from the users storage.
Also make sure when you setup your controller in EmulationStation you configure while it's in wireless mode if you have one.

shantigilbert, 2021-09-06:
> Hey, sorry for Necro but I think the dev branch should have fixed this issue. The problem was gptokeyb was not loading the gamecontrollerdb.txt from the users storage.
> Also make sure when you setup your controller in EmulationStation you configure the keys while it's in wireless mode if you have one.

This has nothing to do with gptokeyb, gptokeyb was not introduced before this issue was present. It's more on the scripts that generate the input files that cannot read the characters.

Langerz82, 2021-09-06:
Ah my bad. Couldnt we just strip the illegal characters for the gamecontrollerdb entry? From my understanding SDL uses the UID and is not phased by the actual title of the device.

shantigilbert, 2021-09-07:
> 
> 
> Ah my bad. Couldnt we just strip the illegal characters for the gamecontrollerdb entry? From my understanding SDL uses the UID and is not phased by the actual title of the device.

The gamecontrollerdb is not the end it all for the controllers in EmuELEC, it is not used by everything (like Retroarch, advmame for example) the issue has nothing to do with SDL (which is where gamecontrollerdb is used) its just the scripts that generate the configs are not prepared for those characters, I am sure its a simple fix, but I have no controller that has a weird name and I do not really like to fix things blindly and wait for testing, etc. So someone with a controller that has those characters will have to fix this one.

Langerz82, 2022-04-19:
Is this still an issue? gptokeyb to kill emulator apps source changed a bit so it might work. If not in SSH type in sdljoytest and press the hotkey and start button and post the results here.

shantigilbert, 2024-01-19:
Will close this since there has not been any response since 2022, reopen if the issue is still present.

closed 2024-01-19 by shantigilbert