EmuELEC/EmuELEC #186: GPL license violation—noncommercial usage

tradej, 2020-05-25:
Hi, I would like to point out that the restriction you put on distributing EmuELEC (no commercial use) is a violation of GPLv2 under which CoreELEC—software on which EmuELEC is based—is distributed under, more specifically sections 1 and 2:

_You may copy and distribute verbatim copies of the Program's source code as you receive it, in any medium..._

and

_You may modify your copy or copies of the Program or any portion of it, thus forming a work based on the Program, and copy and distribute such modifications or work under the terms of Section 1..._

and section 6:

_You may not impose any further restrictions on the recipients' exercise of the rights granted herein..._

For that reason, I request you remove the restriction of commercial distribution. Thank you.

shantigilbert, 2020-05-25:
No, I will not. The code I provide is released under the GPLv2+ and as such and as you noted can be used under the GPLv2+ for whatever the license allows. However what is non-commercial are many emulation cores, thus you cannot sell the resulting image, EmuELEC is not a SINGLE program, is a combination of many (a distro), each one of them with their own licenses. 

refer to https://github.com/libretro/Lakka-LibreELEC/issues/389

If you really want to sell EmuELEC you will need to remove each and every non-commercial core and emulator, then provide the source for your changes.

Good luck.

tradej, 2020-05-25:
You should make the licensing situation clear in the README or LICENSE file then, as that information seems to be missing entirely from your repository. Let me note that the CoreELEC source code is released under GPLv2 strictly, so GPLv2+ applies only on files that you have written yourself from scratch, and which do not fall under the derivative work scope.

shantigilbert, 2020-05-25:
All the files I make (from scratch) are marked as such.

```
# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 20xx-present Shanti Gilbert (https://github.com/shantigilbert)
```

If that is not clear enough I have no problem including it on the README.

would that make it clear to you?

tradej, 2020-05-25:
> I have no problem including it on the README

Please do. This is a fairly complex licensing scenario, which may make many people confused as to what they actually can or can not do when modifying or redistributing your code. After all, I was confused at the beginning too.

> CoreELEC is also GPL-2.0-or-later

Parts of it, yes. However, CoreELEC [README](https://github.com/CoreELEC/CoreELEC/blob/coreelec-9.2/README.md) says:

```
CoreELEC original code is released under GPLv2.
```

For that reason, all original CoreELEC files that don't have their own licence header (such as the one you've linked) should be treated as GPLv2 strictly.

shantigilbert, 2020-05-25:
> Please do. This is a fairly complex licensing scenario, which may make many people confused as to what they actually can or can not do when modifying or redistributing your code. After all, I was [...]

Yes, but "Let me note that the CoreELEC source code is released under GPLv2 strictly" I was replying to this specifically, it is not "stricktly".

I will update the README.

tradej, 2020-05-25:
> it is not "stricktly"

Ah, I see. What I meant by "strictly" is that those files that do not specify "or later" in their header can not be redistributed under a newer GPL licence. For instance, the Makefile from the CoreELEC repository will only be redistributable under the terms of GPLv2.

tradej, 2020-05-25:
> I will update the README.

Thank you.

shantigilbert, 2020-05-25:
Hopefully this clarifies the situation.

tradej, 2020-05-26:
Thank you very much, that helps tremendously.

I have only one question about the branding section: in spirit of clarity, would you consider distributing EmuELEC logos, videos, etc. under one of the existing Creative Commons licences, e. g. CC BY SA? In your current wording, several points are IMO fairly unclear (e. g. what does "are available" mean—does it mean that the person shares them, or that they can be obtained from any other source? Etc.). The standard wording of CC should take care of that.

shantigilbert, 2020-05-26:
Should be "All EmuELEC related logos, videos, images and branding in general are the sole property of EmuELEC and they are all Copyrighted by the EmuELEC team and are not to be included in any commercial application whatsoever without the proper authorization, (yes, this includes EmuELEC bundled with ROMS for donations!).

You are however granted permission to include/modify them in your forks/projects as long as they are completely open-source, freely available, and do not infringe on any copyright laws, even if you receive donations for such project (we are not against donations for honest people), we only ask that you give us the proper credit and if possible a link to this repo." Edited accordingly

I think its clear enough, at this time I do not want to add yet another license, especially since I am against people bundling EmuELEC alongside a ton of ROMS with our logos and such, share them and then ask for donations. AFAIK CC does not cover that. Also I do not want to later find out there is an obscure clause that will have me yet again wasting my time with this.

shantigilbert, 2020-05-26:
Can this issue be closed now?

tradej, 2020-05-27:
The reason I'm suggesting a standard licence is that it creates legal certainty about what can and can not be done with your code, and is well understood in the open-source world. If you use your own wording, you essentially make it harder for people to be certain and not open to a lawsuit, for instance, and they will require legal counsel if they are serious about it.

Of course, I can only ask you to consider it, so if this is not what you want to do, feel free to close this issue.

shantigilbert, 2020-05-27:
I think it's clear enough

closed 2020-05-27 by shantigilbert