EmuELEC/EmuELEC #189: SD card class 10 not running games

aagobbo, 2020-06-06:
I had a 8gb sd card class 4 that works ok, but I needeed some more space for psp games... then i bought a 64gb class 10 sd card, and it works only psp game, using the very same box (a Z69 4k 3gb / 32gb - S905X).

I change for the old sd card and the same snes, n64, megadrive games works, same rom...

I think that there's something about the class 10, maybe something is happening too fast and the emuelec is getting some trouble to load...

Is there something i can do??

SumavisionQ5, 2020-06-06:
I think there's something wrong with your sd card.
Most of sd cards with good quality run great regardless of speed.
Maybe try to reflash your card?

aagobbo, 2020-06-06:
already did that, reflashed 2 times... but it runs psp games smoothly, so, i think it's not damaged...

SumavisionQ5, 2020-06-06:
Try to run a game that you could not run with you C10 card. Then try to read emuelec.log, you will find something useful there.

SumavisionQ5, 2020-06-06:
You can get more info on how to trace your issue from Wiki.
https://github.com/EmuELEC/EmuELEC/wiki

aagobbo, 2020-06-06:
EmuELEC Run Log
PLATFORM: snes
ROM NAME: /storage/roms/snes/Aladdin (U) [!].smc
BASE ROM NAME: Aladdin (U) [!].smc
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/snes/Aladdin (U) [!].smc
2nd Argument: -Psnes
3rd Argument: --controllers=-p1index 0 -p1guid 030000005e0400008e02000014010000 
4th Argument: 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/snes9x_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/snes/Aladdin (U) [!].smc
Emulator Output is:
[WARN] --libretro argument "/tmp/cores/snes9x_libretro.so" is neither a file nor directory. Ignoring.
[INFO] RetroArch 1.8.5 (Git 761a17ccd8)
[INFO] Redirecting save file to "/storage/roms/snes/Aladdin (U) [!].srm".
[INFO] Redirecting savestate to "/storage/roms/snes/Aladdin (U) [!].state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Apr  1 2020
[INFO] Version: 1.8.5
[INFO] Git: 761a17ccd8
[INFO] =================================================
[ERROR] Frontend is built for dynamic libretro cores, but path is not set. Cannot continue.
[ERROR] Fatal error received in: "init_libretro_symbols()"

SumavisionQ5, 2020-06-06:
It seems the system could not load the core from /tmp/cores/snes9x_libretro.so. 
No idea what's wrong. Maybe your sd card is fake or corrupted?

aagobbo, 2020-06-06:
i found this

https://forums.libretro.com/t/retroarch-1-3-4-libretro-path-config-key-broken-using-appendconfig/4971

aagobbo, 2020-06-06:
it seems something like my problem

aagobbo, 2020-06-06:
SOLVED!!!!!

I was analysing the differences between the sd cards I realized that there was a folder Cores with a bunch of files in the old sd card and the same folder was empty in the new one... I just copied pasted and bingo! Working!!!

closed 2020-06-06 by aagobbo

shantigilbert, 2020-06-06:
While the workaround worked, the issue might be with how you flash the image, as this should not happen at all (I use faster cards than Class 10 and have no problems).

aagobbo, 2020-06-07:
The assets, joypad, overlays and shaders had the same problem

shantigilbert, 2020-06-07:
Yeah, they will fail because the image was not flashed correctly probably (joypads are not included anyways)

aagobbo, 2020-06-09:
so, i found what happened (or at least it was the solution for me). I was trying to reflash the card then i realized that the card was using FAT, I changed do NTFS and now everything is ok, with cores, assets and stuff!

Thanks for the help!!

shantigilbert, 2020-06-09:
Good to know!