EmuELEC/EmuELEC #190: FAILURE: scripts/build linux:host has failed!

fabiohpa, 2020-06-07:
Hi! I need some help, please. I'm a beginner and when I try to build the project I have the error return below:

[image]

He had executed this command:
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image  

Tks.

coach1988, 2020-06-07:
Looks like your toolchain is missing ccache,which should come by default I think.my first steps would be:
1. Just try it again,the build is not 100% threadsave,so perhaps another job didn't finish on time
2. Try to start building that package from scratch by running ./scripts/clean linux (perhaps try it with ccache instead of linux,too) and try it again

fabiohpa, 2020-06-07:
Hi coach88, 

Initially, I cloned the repository then and changed just a few splash files for testing. In addition, I need to increase the size of the vm due to lack of space. Move the director "EmuELEC" to this new partition and run the command again to build the project.

I will follow the steps you suggested. 

Tks

fabiohpa, 2020-06-07:
Hi,

I clone the project again and did the build again but now I had the error below:

[image]

Would there be anything else to be done?

Tks

shantigilbert, 2020-06-07:
You need to scroll back to see the real error

fabiohpa, 2020-06-07:
Tks shantigilbert, I'm posting part of the stack and I realized that the first mistake is in this part.

**[ 55%] Linking C executable curltest
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/toolchain/lib/libssl.so: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status**

Next error:
**make[3]: *** [Utilities/cmcurl/CMakeFiles/curltest.dir/build.make:88: Utilities/cmcurl/curltest] Error 1
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[2]: *** [CMakeFiles/Makefile2:1533: Utilities/cmcurl/CMakeFiles/curltest.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[ 86%] Built target CMakeLib
make[2]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[1]: *** [Makefile:163: all] Error 2
make[1]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'**

Part of the stack
gmake[1]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu/Bootstrap.cmk'
gmake[1]: 'cmake' is up to date.
gmake[1]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu/Bootstrap.cmk'
loading initial cache file /media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu/Bootstrap.cmk/InitialCacheFlags.cmake
qmake: could not exec '/usr/lib/i386-linux-gnu/qt4/bin/qmake': No such file or directory
qmake: could not exec '/usr/lib/i386-linux-gnu/qt4/bin/qmake': No such file or directory
-- Configuring done
-- Generating done
-- Build files have been written to: /media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu
---------------------------------------------
CMake has bootstrapped.  Now run gmake.
Executing (host): make 
make[1]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[2]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[  1%] Built target cmsys_c
[  2%] Built target kwiml_test
[  4%] Built target cmsys
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[  5%] Built target cmlibrhash
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[  5%] Built target cmcompress
[  6%] Built target cmzlib
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[  8%] Built target cmbzip2
[  7%] Built target cmexpat
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Entering directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[  8%] Built target cmjsoncpp
make[3]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[... cut ...]
make[2]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
make[1]: *** [Makefile:163: all] Error 2
make[1]: Leaving directory '/media/fabio/Arquivos/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/cmake-3.14.5/.i686-linux-gnu'
[025/416] [FAIL] build   cmake:host
FAILURE: scripts/build cmake:host has failed!
parallel: This job failed:
package_worker 1 25 416 'build cmake:host'
Parallel build failure - see log for details. Time of failure: Sun Jun  7 13:13:24 -03 2020
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

coach1988, 2020-06-07:
Looks like ask has a problem.try cleaning the libssl package with the clean script.

fabiohpa, 2020-06-07:
Ok. Do I need to execute just the command (./scripts/clean linux) or do I need to enter some other parameter?

I ask because I already executed the command ( (./scripts/clean linux) ) and tried to run again PROJECT = Amlogic ARCH = arm DISTRO = EmuELEC make image. I continue with the same error. :(

I realized that the version of cmake is cmake-3.14.5 and my Ubuntu system is in version 16.04. Do you believe that this may have something to do with the problem?

Tks again.

coach1988, 2020-06-07:
Replace linux with ssl (or maybe libssl) and idk if it really matters,but add the PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC,just in case.

fabiohpa, 2020-06-07:
I tried to run the commands with ssl and libssl and got the error return too.

[image]

I will try to do it from scratch one more time. I will delete the project, perform the clone again, change the image and build files.

Thank you for the patience. It will work at some point. :)

shantigilbert, 2020-06-07:
If you are running a VM any reason why not just update to Ubuntu 18.04.4 LTS ? I am certain that the erros mighht be because your OS is a bit old.

```
sudo apt update && sudo apt upgrade
sudo apt-get install gcc make git unzip wget xz-utils libsdl2-dev libsdl2-mixer-dev libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev rapidjson-dev libasound2-dev libgl1-mesa-dev build-essential libboost-all-dev cmake fonts-droid-fallback libvlc-dev libvlccore-dev vlc-bin texinfo premake4 golang
git clone https://github.com/shantigilbert/EmuELEC.git EmuELEC    
cd EmuELEC  
git checkout master  
```
you can also check out the dev branch.

fabiohpa, 2020-06-07:
A previous version had been installed due to disk space. I had another project that didn't need a newer version of the distro.

About the commands.. They were the same ones that I executed.

At this point I am running the steps again, but without my change.

If it doesn't work I will try to upload the system version.

Tks

british-choi, 2020-06-08:
@fabiohpa
Do you have enough storage space?
Show me the your storage info.

$df -h

fabiohpa, 2020-06-08:
Hi! I now have enough space. I created a new vm with 54gb of storage, installed version 18 of Ubuntu and did the build again. 

It's been running for a few hours. 

I believe it worked! :)

fabiohpa, 2020-06-08:
Even creating a 54gb vm the process was not completed due to lack of space. After all how many gb do I need to build the system?

[image]

fabiohpa, 2020-06-08:
Guys, I will close this topic I believe that the problem is another now. I am grateful to everyone who contributed.

Tks

closed 2020-06-08 by fabiohpa