EmuELEC/EmuELEC #192: FAILURE: scripts/install u-boot has failed!

fabiohpa, 2020-06-08:
Hi guys, when trying to build the project in my vm I got the error below.

[image]

I redid the steps described in the readme and the bug persists.

I agree that I did not understand the reason for the error and then correct it. Could someone help me? I am a beginner and would like to learn how to compile the project and then start to cooperate with ideas.

Tks

Ubuntu 18.04
HD 200 GB
EmuELEC 3.5

shantigilbert, 2020-06-09:
You need to provide the log for that failure, or scroll back a few lines until you see the real error, from that image is impossible to know

british-choi, 2020-06-09:
@fabiohpa 
you can get failed log
$ cat ./build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout

fabiohpa, 2020-06-09:
Try to run or command to export or not work. I chose to copy and paste. Sorry for the number of lines.

Everything was going well, I didn't understand what could have happened.

Thank you for your help.

`fabio@fabio-VirtualBox:~/EmuELEC$ PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image
./scripts/image mkimage

 =================================================================================
 Configuration for EmuELEC (community)
 =================================================================================

 Buildoptions:
 ======================================================
 - CPU (ARCH):				 cortex-a53 (arm)
 - FLOAT:				 hard
 - FPU:					 neon-fp-armv8
 - CPU features:			 32bit neon
 - LTO (Link Time Optimization) support: yes
 - GOLD (Google Linker) Support:	 yes
 - LLVM support:			 no
 - DEBUG:				 no

 Graphic configuration:
 ======================================================
 - OpenGL (GLX) support (provider):	 no (no)
 - OpenGLES support (provider):		 yes (opengl-meson-t82x)
 - uvesafb support:			 no

 Hardware decoder configuration:
 ======================================================
 - Kodi Player driver:			 libamcodec
 - VAAPI Support:			 no
 - VDPAU Support:			 no

 Input device configuration:
 ======================================================
 - Remote support:			 no
 - CEC Adapter support:			 no
 - CEC Framework support:		 no
 - Kodi Joystick support:		 yes

 Misc. hardware configuration:
 ======================================================
 - ALSA support:			 yes
 - Pulseaudio support:			 yes
 - Bluetooth support:			 yes
 - Include driver:			 gpu-aml
 - Include driver:			 kvimfan-aml
 - Include driver:			 openvfd-driver
 - Include driver:			 wifi_dummy-aml
 - Include driver:			 ap6xxx-aml
 - Include driver:			 mt7601u-aml
 - Include driver:			 mt7603u-aml
 - Include driver:			 mt7610u-aml
 - Include driver:			 qca9377-aml
 - Include driver:			 ssv6xxx-aml
 - Include driver:			 RTL8188EU-aml
 - Include driver:			 RTL8189ES-aml
 - Include driver:			 RTL8189FS-aml
 - Include driver:			 RTL8192CU
 - Include driver:			 RTL8192DU
 - Include driver:			 RTL8192EU
 - Include driver:			 RTL8723BS-aml
 - Include driver:			 RTL8723DS-aml
 - Include driver:			 RTL8812AU
 - Include driver:			 RTL8814AU
 - Include driver:			 RTL8821CU
 - Include driver:			 RTL8822BU-aml
[... 228 lines cut ...]
INSTALL      gcc-linaro-aarch64-elf (host)
INSTALL      gcc-linaro-arm-eabi (host)
INSTALL      u-boot-LePotato (target)
[104/416] [DONE] install u-boot-LePotato
INSTALL      u-boot_firmware (target)
[105/416] [DONE] install u-boot_firmware
INSTALL      u-boot-Odroid_C2 (target)
[106/416] [DONE] install u-boot-Odroid_C2
BUILD      u-boot (target)
    TOOLCHAIN      manual
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/update.sh' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/update.sh'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/canupdate.sh' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/canupdate.sh'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/LePotato_boot.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/LePotato_boot.ini'
'/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-LePotato-b497a82d864d07e4d4e8ffe7f05870536b2b38d0/fip/u-boot.bin.sd.bin' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/LePotato_u-boot'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/Odroid_C2_boot.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/Odroid_C2_boot.ini'
'/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-Odroid_C2-ad3e8b1cfb74d22777cb002f01ac992abe457aad/u-boot.bin' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/Odroid_C2_u-boot'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/config.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/config.ini'
[107/416] [FAIL] install u-boot
FAILURE: scripts/install u-boot has failed!
parallel: This job failed:
package_worker 1 107 416 'install u-boot'
Parallel build failure - see log for details. Time of failure: Mon Jun  8 22:46:34 -03 2020
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1
`

british-choi, 2020-06-09:
1. $ cat ./build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout
2. $ df -h

fabiohpa, 2020-06-09:
@british-choi , I try to execute the command but it returns the message that the path does not exist. 

I enabled the viewing of hidden files and accessed the threads folder. There are several files, I noticed that there is a file called "u-boot: target.build.failed" and inside it is written "187". 

It helps?

[image]

Tks

shantigilbert, 2020-06-09:
This is a very weird issue as there is really no error message, it seems everything has been copied as it should, are you sure yoiu have enough space? 

run `df -h` 

also make sure your directories are NOT symlinks.

british-choi, 2020-06-09:
your capture image is "107" not "187"
$ cat ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout
$ df -h

fabiohpa, 2020-06-09:
@british-choi, sorry, i just typed wrong. Had copied and pasted the commands you gave me.
$ cat ./build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout

@british-choi I ran the command cat ~ / EmuELEC / build.EmuELEC-Amlogic.arm-3.5 / .threads / logs / 107 / stdout and got the same return.
[image]

@shantigilbert, I ran the command and got the following return...
[image]

shantigilbert, 2020-06-09:
can you try this 

`PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC ./scripts/build u-boot`

fabiohpa, 2020-06-09:
@shantigilbert I ran the command

[image]

shantigilbert, 2020-06-09:
ok, now try the regular  command
 `PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image`

fabiohpa, 2020-06-09:
@shantigilbert  same mistake

[image]

shantigilbert, 2020-06-09:
hmm this is extremely weird... 

Try this now 
`PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC ./scripts/install u-boot`

fabiohpa, 2020-06-09:
@shantigilbert,  it's insane .. hahaha

I ran the command

[image]

fabiohpa, 2020-06-09:
@shantigilbert 

tried again regular command
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image

[image]

british-choi, 2020-06-09:
what directory (or file) is missing?
1. ~/EmuELEC
2. ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5
3. ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads
4. ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs
5. ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107
6. ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout  <= this is not directory, It's log file

shantigilbert, 2020-06-09:
@fabiohpa You need to provide the file that @british-choi is asking for, as the error is on package 107, whichever that might be, it is not u-boot

also you can provide `~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stderr`

fabiohpa, 2020-06-09:
@shantigilbert  @british-choi  there is no "logs" folder on that path you said.

[image]

british-choi, 2020-06-09:
sorry, i don't know this issue.
I'm not use virtual box system
The only difference from me is that you are using a virtual box.

shantigilbert, 2020-06-09:
I am not sure what the issue is either, I also do not use VM. but try this 

`find . -type d -name "107"` 

maybe the logs are set to be saved elsewhere with the VM.

fabiohpa, 2020-06-11:
Hi guys, i'm back. hahaha

I installed Ubuntu 18.04 on a new computer. I'm not using vm anymore. I tried to compile the project again and got the same error but I was able to access the logs. Accessing the stderr file from the logs folder 107. I got the content to follow.

[108/416] [FAIL] install u-boot
The following logs for this failure are available:
  stdout: /home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stdout
  stderr: /home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/107/stderr

Running the command:
"$ cat ./build.EmuELEC-Amlogic.arm-3.5/.threads/logs/108/stdout" . I got the content to follow.

(...) 
libbb/xfuncs_printf.c: In function 'generate_uuid':
libbb/xfuncs_printf.c:649:3: warning: ignoring return value of 'read', declared with attribute warn_unused_result [-Wunused-result]
   read(i, buf, 16);
   ^~~~~~~~~~~~~~~~
make[2]: *** wait: No child processes.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** wait: No child processes.  Stop.
make[1]: *** [Makefile:743: libbb] Error 2

shantigilbert, 2020-06-11:
can you post the complete error log, not just the end. maybe use a site like pastebin

fabiohpa, 2020-06-11:
hi @shantigilbert, I couldn't embed the pastebin link. I will make the link available:

https://pastebin.com/VUy0dBVs

shantigilbert, 2020-06-11:
> hi @shantigilbert, I couldn't embed the pastebin link. I will make the link available:
> 
> https://pastebin.com/VUy0dBVs

I really don't know what is going on, these are my suggestions

try this first

```
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC ./scripts/build busybox
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC ./scripts/install busybox
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image
```
you might get a clearer error, although I doubt it 

if that does not work, try this:
```
git checkout dev
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC ./scripts/build busybox
PROJECT=Amlogic ARCH=arm DISTRO=EmuELEC make image
```

fabiohpa, 2020-06-11:
cool, I'll try again and post the result.

tks @shantigilbert

fabiohpa, 2020-06-12:
@shantigilbert.. i perform the previous steps and I was unsuccessful. What a shame!
I ran the commands to validate the logs and the return is below.

What version of linux do you use?

fabio@fabio-HP-G42-Notebook-PC:~/EmuELEC$ cat ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/108/stderr
[137/419] [FAIL] install u-boot

The following logs for this failure are available:
  stdout: /home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/108/stdout
  stderr: /home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/108/stderr

fabio@fabio-HP-G42-Notebook-PC:~/EmuELEC$ cat ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/108/stdout
BUILD      u-boot (target)
    TOOLCHAIN      manual
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/update.sh' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/update.sh'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/canupdate.sh' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/canupdate.sh'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/LePotato_boot.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/LePotato_boot.ini'
'/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-LePotato-923e6512c88d595df9514a719833c72933d6d2bd/fip/u-boot.bin.sd.bin' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/LePotato_u-boot'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/Odroid_C2_boot.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/Odroid_C2_boot.ini'
'/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-Odroid_C2-a2eb29f823e8cfc6fee325f69116334c6f1ba6c0/u-boot.bin' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/Odroid_C2_u-boot'
'/home/fabio/EmuELEC/projects/Amlogic/bootloader/config.ini' -> '/home/fabio/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/u-boot-1.0/.install_pkg/usr/share/bootloader/config.ini'
FAILURE: scripts/install u-boot has failed!

fabio@fabio-HP-G42-Notebook-PC:~/EmuELEC$ cat ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/137/stdout
INSTALL      busybox (host)
INSTALL      busybox (target)

fabio@fabio-HP-G42-Notebook-PC:~/EmuELEC$ cat ~/EmuELEC/build.EmuELEC-Amlogic.arm-3.5/.threads/logs/137/stderr
[134/419] [DONE] install busybox

tks

shantigilbert, 2020-06-12:
Ubuntu Ubuntu 18.04.4 LTS 64bits

I really do now know what the error is, sorry.

fabiohpa, 2020-06-12:
@shantigilbert 

Thank you for the support. 👍

closed 2020-06-12 by fabiohpa