EmuELEC/EmuELEC #196: Future support for 4:3 aspect ratio screens?

shateredsoul, 2020-06-15:
I'm loving Emelec. Thanks for the work.  I use it on a 4:3 screen on a mini arcade I built. I currently have to set the screen size to 16:9 otherwise it looks squished since it's set up for a widescreen monitor. Any chance that there may be support for 4:3 screens in the future?

shantigilbert, 2020-06-16:
If you mean ES supporting 4:3 then you need a theme that can do it, if you mean emulators supporting  4:3 this can be done on the game settings in ES

closed 2020-06-16 by shantigilbert

dmb062082, 2021-02-21:
> I'm loving Emelec. Thanks for the work. I use it on a 4:3 screen on a mini arcade I built. I currently have to set the screen size to 16:9 otherwise it looks squished since it's set up for a widescreen monitor. Any chance that there may be support for 4:3 screens in the future?

ShateredSoul I know this is off topic but I have been trying to find you for a few weeks. This seems to be the only way, and I do not see a PM option. I had a few questions for you. My email is dmb062082@gmail.com or you can text me at 680.219.2596 Its not urgent.

David