EmuELEC/EmuELEC #199: [OGA] Retroarch freezing on exit when auto save/load enabled

Desani, 2020-06-17:
**Describe the bug**
Under game settings enabling AUTO SAVE/LOAD to ON and then launching a game causes RetroArch to freeze on game exit when attempting to save state.

**To Reproduce**
Steps to reproduce the behavior:
1. Set AUTO SAVE/LOAD to ON under Game Settings in Emuelec
2. Start a game (Tested SNES, NES, GBA, Megadrive)
3. Play the game and then exit back to Emuelec, it might freeze and become unresponsive.

**EmuELEC version:**
 - Version 3.6

**Device (please complete the following information):**
 - Model Odorid Go Advance

**Additional context**
I killed the RetroArch Process ID through SSH and then it returns to normal operation after the process has been killed twice.

shantigilbert, 2020-06-17:
Disable save compression and try again, this is a RA issue not exclusive to EmuELEC.

Desani, 2020-06-17:
I have disabled save compression and it continues to be an issue.

save_file_compression = "false"

I did not have this issue with 3.5 or 3.6 Test and I did not have to change the compression setting in previous releases.

shantigilbert, 2020-06-17:
Like I said, this is a RA issue, you didn't had it because RA was a different version then, if that didn't work there is nothing I can really do other than wait for a fix on RA

Desani, 2020-06-17:
Okay sounds good. I will just leave AUTO SAVE/LOAD disabled for the time being and wait for an update from that end.

closed 2020-06-20 by Desani