EmuELEC/EmuELEC #206: S905x3 not getting any sound from audio out

Treyarch, 2020-06-21:
Hello,

Just configured my  S905x3 with EmuELEC but I'm not getting any sound from the audio jack. I'm getting sound through HDMI but I'm setting up a my Bartop Arcade and I want sound out from small speakers connected to an amplifier and audio jack out.

I've tried to chose the audio device (0,0 through to 1,1) from the EmuELEC settings but with no success, any help is appreciated !

Thanks !

farbirdsg, 2020-06-21:
Try to ssh into the box and run alsamixer.

Treyarch, 2020-06-22:
Whoa, this is nice, I will try to fiddle around a bit and figure it out, thanks !

Treyarch, 2020-06-22:
Got a question though, how do I know which is which is the settings ? Is there a term specific to audio jack out for all mixers ?

Treyarch, 2020-06-22:
ok, got no idea how to get this to work. I'm posting captures if anyone can help me out figuring this !

https://imgur.com/a/sHkP0wf

Treyarch, 2020-06-22:
Ok, I'm referencing this [post](https://discourse.coreelec.org/t/emuelec-in-x96-air-3-5mm-headphone-output/10328) as it seems there is no fix for now, audio out through 3.5mm jack is not working :(

shantigilbert, 2020-06-22:
> I've tried to chose the audio device (0,0 through to 1,1) from the EmuELEC settings but with no success, any help is appreciated !

When you apply these settings, do you press B (or back) and then reboot the machine?

Treyarch, 2020-06-23:
@shantigilbert Yes I pressed Back and I tried rebooting but it changes nothing, audio is not working in the CoreElec build for now.

shantigilbert, 2020-06-24:
Hmm I just tested this with a Beelink s905x3 device running v3.6 and I can get sound from the analog (0,1) as well as HDMI (0,0) 
I just selected 0,1 on the Main Menu > EmueLEC settings > Audio Device press B to go back and save the selection and reboot (rebooting after the selection is important). 
 if you have HDMI connected you will get audio from both, I do not know how to fix that or if that is even considered a bug.

DellendinMi, 2020-06-26:
@Treyarch
As @shantigilbert described, i can confirm that analog audio from AV output port is working with my X96 Air s905x3 on EmuELEC v3.6.

The coreelec post you gave the link above is from April, 2 months ago. With the newest CoreELEC 9.2.3 (June) version it is stated that: "Fixed analog audio on S905X3 devices". Since Emuelec v3.6 is "Synched with CoreELEC 9.2.3", then it's obvious that we can expect a working analog audio output.

Let me give the details of my experience, maybe it can help someone somehow:

I had been using the previous EmuELEC v3.5 on my X96 Air 4/64GB S905X3 tv box with HDMI 1080p output for 3 weeks. Thanks to @shantigilbert for his great work, by the way. Then i wanted to try the 576cvbs AV analog output, because i plan to use this Tv Box with a Philips Flat CRT Tv, in the end. When i connected to TV with AV analog output, i was able to get video but NO AUDIO. I made a research on the web, and saw that the newest CoreELEC 9.2.3 (June) version fixed analog audio on S905X3 devices. So, i decided to upgrade Emuelec v3.5 to v3.6, because @shantigilbert states that: "Synched with CoreELEC 9.2.3".

I didn't upgrade, but made a clean install of Emuelec v3.6, although it costed me more time because of backups and reloads (I feel more comfortable with clean installs.) With a great excitement, i tried 576cvbs AV analog output. As i expected i couldn't get the audio again first with the default "Audio Device" > Auto setting. Then i changed it to (0,1) and restarted the box. At first while hovering on the menu entries and scraper videos, i couldn't hear any audio. Then i started a PSX game, and realized that there's audio while the game is running. Since the sound was very low, i nearly missed out the audio playing. I realized that; with the exact same TV and Emuelec sound levels, the audio output of AV is much lower than with the HDMI. So be aware of this, although this may differ from TV to TV, box to box. So you'd better think of increasing the volume levels in every try you make.

As i explained above, i still did not hear any "tik, tik, .." sounds on the system menu entries and with the scraper videos. (There's no chance to miss out these sounds, because they are very high than the games and i couldn't find a way to lower these sounds. Especially in the scraper videos, so i prefer to deactivate "video sounds" to mute scraper videos.) When i made the second restart of the box, these sounds became available. So, it may be necessary to make **2 reboots** to get full audio on games, menus, and scraper videos. I'm not sure, it could be just my luck/experience.

In the last step, i changed the output to HDMI 1080p with the "Audio Device" --> (0,1) setting unchanged. As stated above i was able to get sound from both HDMI TV speakers, and the headphone connected to AV analog output. I would expect that this situation could be possible with a "Audio Device" --> (1,1) setting than a (0,1) setting, though.

Long story short, i'm glad to know that my s905X3 Tv Box is working perfectly with AV output as well. Thanks again to @shantigilbert and all contributers. Happy retro gaming to all!

shantigilbert, 2020-06-28:
Closing this as it has been confirmed working on several devices.

closed 2020-06-28 by shantigilbert