EmuELEC/EmuELEC #210: X96 mini p281 and p282

chenteparao, 2020-06-27:
Hi, I bought a x96mini some days ago and performance is amazing with emuelec, today bought the "same" device for my brother but not boot. I install AIDA64 in both devices and old X96mini has a difference, 

it say p281 in board, while new device say p282 in board. Do you know how to run in this new device? Thanks.

[image]
[image]

chenteparao, 2020-06-30:
[image]
With terminal shoe that

George2912, 2020-11-08:
[image]
help, is this x96mini clone? 

sorry for my english i use a translator

closed 2021-04-01 by shantigilbert