EmuELEC/EmuELEC #213: Emuelec 3.6 usd disconnect

javis, 2020-07-02:
**Describe the bug**
After updating to the new version, my usb controllers started to disconnect randomly, as if were consuming too much power (if I disconnect one, the other starts working). This did not happen with the previous version. I am using Odroid N2

**To Reproduce**

1. Start the device in Emuelec with 2 USD controllers connected. 
2. You will see screen saying there were not controllers detected
3. Disconnect one gamepad
4. Restart
5. Controller is detected. 

EmuELEC version:
Version 3.6

Device (please complete the following information):

SOC: s922x
Brand: odroid
Model: odroid N2
Version

**Additional context**
This happened to me with Batocera, but when I switched to Emuelec 3.5 never happened again, so I decided to stick with Emuelec for that reason. After updating version it started to happen again. Some posts on Odroid forums says it is a driver version on linux that needs to be updated.

shantigilbert, 2020-07-03:
have you tried other USB ports? and which driver are you referring to?

javis, 2020-07-03:
Yes I have tried will all the USB ports, it's the same for all.
I am referring to the USB drivers, it seems it is a common problem among
many different Odroid models

https://forum.odroid.com/viewtopic.php?t=34849

shantigilbert, 2020-07-03:
Unfortunately if that is in fact an issue with the HW I am not sure I can do much... they do not seem to have a solution on that thread, can you install 3.5 on another SD and see if the problem is in act not present there?

javis, 2020-07-03:
Ok I will go back to 3.5 and let you know

javis, 2020-07-04:
Ok I have tried with 3.5 and the problem is not present. But I have found in internet that is a common problem with the Zero Delay USB interfaces, I don't know why it was working on 3.5 in the first place. Maybe the Steam Controller detection had something to do?

shantigilbert, 2020-07-04:
Let me investigate and I will hg back at you

shantigilbert, 2020-07-04:
I unfortunately do not have but one zero delay USB adapter, but I connected 3 gamepads (tested it weith these controllers in different configurations: Logitech f310, Xbox 360, Hori Switch controller and zero delay USB arcade stick that identifes itself as dragonrise) + external HDD (no external power) on my n2 running 3.6  and I see no issues. I started it with all devices connected/disconnected and it works fine. Maybe you can provide dmesg logs?

javis, 2020-07-04:
hum. yes this problem occurs with 2 of them connected at the same time.
although it might have micro disconnects during gameplay with just one
plugged in. I will see if I can get the logs and do more investigation on
the issue.

shantigilbert, 2020-11-26:
Is this still happening? I have 2 zero delay adapters now, and it seems for me its working fine.

javis, 2020-11-26:
Turns out it was interference from other electronic apparels, I guess it
has to do with the fact that I'm using boards without any case or kind of
isolation

shantigilbert, 2020-11-26:
Weird, anyways, glad its fixed.

closed 2020-11-26 by shantigilbert

durstboiler, 2021-07-07:
After researching the problem for a long time, I finally found the source. This is a power supply interference. I moved the power supply away from other power supplies and wiring, and instantly no more problems.