EmuELEC/EmuELEC #214: Scraping and trying to start a game at the same time resulted in a completely damaged SD Card (OGA BE)

ouroboros420, 2020-07-03:
Odroid Go Advance Black Edition User on EmuELEC 3.6 here. Started a scraping process and wanted to see, if it is possible to play a game at the same time. After launching the game, emuelec tried to restart itself, but couldn't. Manually powering off and on again resulted in black screen after boot. Tried to reflash the OS Image, but the SD card wasn't able to be mounted anymore. Till now I haven't found a way to format the SD card, it seems like it is broken beyond repair. Writing this post to warn people that they could break their sd card when one tries to scrape and play a game at the same time. Now I gotta purchase a new sd card. Oof.

'sudo dmesg' showed me, that the sd card has "I/O error", which basically means, that the SD card is broken:

"I/O error from an SD card means that it's dead. That's why no software lets you format it. The card isn't able to store data any more. Some of its area might still be usable, but it's missing some of the bits. [...] You can't fix an SD card. Throw the SD card away and use a working one." (Source: https://unix.stackexchange.com/questions/236252/how-to-fix-this-i-o-error-on-a-sd-card)

shantigilbert, 2020-07-03:
This is probably just a coincidence, there is no way the SD got killed by that, I've ran games while scraping before, the scrape just stops, it will not kill an SD card.

Either there was already a problem with your SD card or there is a other issue at hand, but if writing data to your SD kills it then emuelec would not work at all.

ouroboros420, 2020-07-03:
Thats nice to hear! It must have been a mixture of my sd card being "fake" (as crashoverride puts it) and having way too less space. It might have been, that the scraping exceeded the space my sd card had. While normal sd cards don't have problems with that, fake sd cards might break from it. 
thanks for your response. I guess you can close this issue.

NeonLightning, 2020-07-03:
i use hw2test in windows to check my sd cards to make sure they'll work right if i get cheap ones. might be worth a shot on future cards.

shantigilbert, 2020-07-03:
I can't assure anything for "fake" cards, I assume if it broke from writing some data to it, It would eventually break on any device.

closed 2020-07-04 by shantigilbert