EmuELEC/EmuELEC #216: Image flickers

b0b33140, 2020-07-05:
**Describe the bug**
My screen constantly flickers for about 2s.

**EmuELEC version:**
EMUELEC 3.6 NG

**Device (please complete the following information):**
 - SOC: S905x2
 - Brand: Beelink
 - Model GT1 mini 4G
 - Version 95X2

ws02589111, 2020-07-09:
I get the same problem, but the flashing frequency faster than 2s

EmuELEC version:
EMUELEC 3.6

Device:

SOC: S905M2
Brand: Q5
Model: None
Version: None

shantigilbert, 2020-07-09:
Does it work when using CoreELEC? or regular Android?  this does not seem related to EmuELEC at all, I have a GT1 mini 4G and I can assure you, I do not get any flickering.

b0b33140, 2020-07-10:
Je n'ai pas ce problème sous Android.
Je vais esayer de remètre la version antérieur (3.5).

b0b33140, 2020-07-12:
Version 3.5 does not have this problem.
I think this problem may not be visible depending on the screen used.

kyo750904, 2020-07-17:
My device also flashes
GT KING S922X

shantigilbert, 2020-11-26:
I cannot reproduce this, sorry.  I will close this issue but feel free to reopen with more information if this issue is still present in newer versions.

closed 2020-11-26 by shantigilbert