EmuELEC/EmuELEC #221: FBA:  "game ratio" and "integer scale" options ignored

caseyjmorris, 2020-07-12:
**Describe the bug**
FBA:  "game ratio" and "integer scale" options ignored

**To Reproduce**
Steps to reproduce the behavior:
0. Start EmulationStation
1. Go to FBA game list
2. Press select while highlighting a game
3. Go to "advanced game options" or "advanced system options" and change the "integer scale" and "game ratio" options.
4. The setting has no effect.  You change change the options in-game by pressing hotkey + X to bring up the RetroArch menu and changing the options there, but this setting will be lost if you exit the game.

**Expected behavior**
Expected the setting to affect the scaling and aspect ratio game is rendered in.

**Screenshots**
(None)

**EmuELEC version:**
 - Version 3.6

**Device (please complete the following information):**
 - SOC: Amlogic S905X3
 - Brand: EVER EXPRESS
 - Model T95 MAX+
 - Version (not aware of multiple versions of this device)

**Additional context**
(None)

shantigilbert, 2020-07-13:
I cannot reproduce this, but first of all, EmuELEC does not have a FBA core not sure if you mean FBneo, but the scale and ratio work fine in FBneo, I just tested it with a handful of games.

caseyjmorris, 2020-07-14:
Yes, sorry, didn't realize there was a difference. 

Settings:

[image]

In game:
[image]

In game menu:

[image]

As you can see, if I change the setting from the in game menu I see the expected result in the background:

[image]

Am I doing something wrong here?

shantigilbert, 2020-07-14:
You never mentioned you use Bezels, disable them to use those options

caseyjmorris, 2020-07-14:
Didn't realize that was an issue, since I just left that setting with its default value.  That seems to fix it.

closed 2020-07-14 by caseyjmorris

shantigilbert, 2020-07-14:
Bezels need predefined aspect or else they will display incorrectly :)