EmuELEC/EmuELEC #222: RetroAchievements don't work with Nintendo 64 ROMs

TheLastProject, 2020-07-12:
**Describe the bug**
RetroAchievements never activates in Nintendo 64 ROMs, making it impossible to get achievements there, no matter the emulator

**To Reproduce**
Steps to reproduce the behavior:
1. Log into RetroAchievements
2. Enabled verbose mode
3. Start a N64 ROM in Mupen64Plus-Next or ParaLLEl
4. Notice there is no "RetroAchievements: Logged in as <yourusername>" message in the bottom left

**Expected behavior**
There should be a successful login when using either Mupen64Plus-Next or ParaLLEl because they are listed on https://docs.libretro.com/guides/retroachievements/#nintendo-64 as supported. The lack of successful login also means no achievements can be gotten

**Screenshots**
Please note, the following screenshot is of Tobu Tobu Girl on GameBoy, which is a free download from https://tangramgames.dk/tobutobugirl/. This screenshot is to show how it SHOULD look when the game starts. On the Nintendo 64, none of these messages show up:
[image]

**EmuELEC version:**
 - Version 3.6

**Device (please complete the following information):**
 - SOC: RockChip RK3326 (I believe, it's not modified)
 - Brand: Odroid
 - Model: Go Advance
 - Version 1.1

**Additional context**
I tried with my copy of Super Mario 64 (PAL) with md5 checksum 45676429ef6b90e65b517129b700308e. This is listed as a known and correct MD5 checksum, see https://retroachievements.org/linkedhashes.php?g=10003

shantigilbert, 2020-07-12:
v3.6 does not have RA enabled for n64   (or other systems like PSX)

https://github.com/EmuELEC/EmuELEC/blob/master/packages/sx05re/emuelec/config/emuelec/scripts/setsettings.sh

but they have been enabled for the upcoming v3.7

https://github.com/EmuELEC/EmuELEC/commit/9a4a8175c01ba7c38305c25f724b1f71ced9ee61

closed 2020-07-12 by shantigilbert

TheLastProject, 2020-07-12:
Is there ever something you haven't fixed already? You're awesome and thank you for the explanation and code links so I can start better understanding EmuELEC too :) 3.7 hype!

shantigilbert, 2020-07-12:
> 
> 
> Is there ever something you haven't fixed already? You're awesome and thank you for the explanation and code links so I can start better understanding EmuELEC too :) 3.7 hype!

There are occasional test (beta ) releases on discord #test-releases channel that you might be interested if you like to tinker around :)