EmuELEC/EmuELEC #228: Emuelec 3.6 addon compiling error

meeuuh92, 2020-07-15:
I'm trying to compile emuelec addon ( for coreeelc) on my intel nuc under debian buster but compiliation failed at retroarch step (error).
I don't know if it' s a bug or not because only 3.2 version is availablebon github.

Regards,

shantigilbert, 2020-07-16:
No idea what error you get, but compilation is only tested on Ubuntu 18.04, any other distro/os is not officially supported.

meeuuh92, 2020-07-16:
Ok... I read a little the logfile... It was some missed dependencies. But now, i've got glib2 support not found... But it's installed (with dev...)  Any idea ?

tarvip, 2020-08-30:
I also tried to build on Ubuntu 18.04 and it also fails building pulseaudio with glib 2.0 not found:
```
...
Installing /home/tarvi/src/EmuELEC/build.EmuELEC-Amlogic.arm-3.7/glib-2.61.1/glib.supp to /home/tarvi/src/EmuELEC/build.EmuELEC-Amlogic.arm-3.7/toolchain/share/glib-2.0/valgrind
Running custom install script '/home/tarvi/src/EmuELEC/build.EmuELEC-Amlogic.arm-3.7/toolchain/bin/meson --internal gettext install --subdir=po --localedir=share/locale --pkgname=glib20'
BUILD      pulseaudio (target)
    TOOLCHAIN      configure (auto-detect)
Executing (target): /home/tarvi/src/EmuELEC/build.EmuELEC-Amlogic.arm-3.7/pulseaudio-12.2/configure --host=armv8a-libreelec-linux-gnueabi --build=x86_64-linux-gnu --prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --disable-static --enable-shared --disable-silent-rules --disable-nls --enable-largefile --disable-rpath --enable-neon-opt --disable-x11 --disable-tests --disable-samplerate --disable-oss-output --disable-oss-wrapper --disable-coreaudio-output --enable-alsa --disable-esound --disable-solaris --disable-waveout --enable-glib2 --disable-gtk3 --disable-gconf --disable-avahi --disable-jack --disable-asyncns --disable-tcpwrap --disable-lirc --enable-dbus --disable-bluez4 --disable-bluez5 --disable-bluez5-ofono-headset --disable-bluez5-native-headset --enable-udev --with-udev-rules-dir=/usr/lib/udev/rules.d --disable-hal-compat --enable-ipv6 --enable-openssl --disable-orc --disable-manpages --disable-per-user-esound-socket --disable-default-build-tests --disable-legacy-database-entry-format --with-system-user=root --with-system-group=root --with-access-group=root --without-caps --without-fftw --without-speex --with-soxr --with-module-dir=/usr/lib/pulse
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk

...

checking for GLIB20... no
configure: error: *** GLib 2 support not found

```

I suspect it is searching glib-2.0 in wrong location `/usr/lib`, but it glib-2.0 was installed to `${TOOLCHAIN}/lib`.

shantigilbert, 2020-08-30:
What command are you using to compile?

tarvip, 2020-08-30:
I am using `./emuelec-addon.sh`. I got pulseaudio actually working with following change:
```
diff --git a/packages/audio/pulseaudio/package.mk b/packages/audio/pulseaudio/package.mk
index b1ef44dd11..63206f0056 100644
--- a/packages/audio/pulseaudio/package.mk
+++ b/packages/audio/pulseaudio/package.mk
@@ -9,7 +9,7 @@ PKG_SHA256="809668ffc296043779c984f53461c2b3987a45b7a25eb2f0a1d11d9f23ba4055"
 PKG_LICENSE="GPL"
 PKG_SITE="http://pulseaudio.org/"
 PKG_URL="http://www.freedesktop.org/software/pulseaudio/releases/$PKG_NAME-$PKG_VERSION.tar.xz"
-PKG_DEPENDS_TARGET="toolchain alsa-lib dbus libcap libsndfile libtool openssl soxr systemd glib:host"
+PKG_DEPENDS_TARGET="toolchain alsa-lib dbus libcap libsndfile libtool openssl soxr systemd glib glib:host"
 PKG_LONGDESC="PulseAudio is a sound system for POSIX OSes, meaning that it is a proxy for your sound applications."
 PKG_BUILD_FLAGS="+pic -lto"

```
But there are other issues as well

shantigilbert, 2020-08-30:
Can you first do a clean build of the system? If you have the time and disk space

shantigilbert, 2020-08-30:
`PROJECT=Amlogic-ng make image `

tarvip, 2020-08-30:
Ok, will try, I thought that this is not needed if building addon only.

shantigilbert, 2020-08-30:
> 
> 
> Ok, will try, I thought that this is not needed if building addon only.

Unfortunately the files are gathered from the build, the addon should build what is needed, but since I never build it like that, I am not sure if this generates the issues.

tarvip, 2020-08-30:
Just to confirm, I have S905, which should be just `PROJECT=Amlogic` or should I still use `Amlogic-ng`.

shantigilbert, 2020-08-30:
> 
> 
> Just to confirm, I have S905, which should be just `PROJECT=Amlogic` or should I still use `Amlogic-ng`.

Yes should be `PROJECT=Amlogic`

The fix you posted is actually the correct one, the same might be needed to be done to other packages that fail because of missing dependencies.

tarvip, 2020-08-30:
Ok, will try.

meeuuh92, 2020-08-31:
On my side, i spent some times to solve missing dependencies....and finally i was stuck with an  error about "glib2  missing" (but the lib and dev is present). Regards,

shantigilbert, 2020-08-31:
> On my side, i spent some times to solve missing dependencies....and finally i was stuck with an error about "glib2 missing" (but the lib and dev is present). Regards,

@meeuuh92 Didn't the fix posted by @tarvip help? seems to be the same problem.

Otherwise without logs is hard to guess.

meeuuh92, 2020-08-31:
I seen on the comments "make image" is it essential to do this step ?  ... I think not and I ve only did emuelecaddon.sh. i will post log tomorrow ? ... (Anyway i test the released version on my c4 and see a video on odroid go advance... A
This is a very nice work..)

shantigilbert, 2020-08-31:
> I seen on the comments "make image" is it essential to do this step ? ... I think not and I ve only did emuelecaddon.sh. i will post log tomorrow ? ... (Anyway i test the released version on my c4 and see a video on odroid go advance... A
> This is a very nice work..)

no, please read this reply: https://github.com/EmuELEC/EmuELEC/issues/228#issuecomment-683454362

meeuuh92, 2020-08-31:
Ah ok... I read the comments too fast.. I will test the diff on pulseaudio pkg.

tarvip, 2020-09-01:
I was able to compile `dev` branch for `Amlogic`, will test it in few hours on a S905 device, if it works, I will create a PR.
For some reason this `pulseaudio` fix was not needed in `dev` branch, had to do only 3 small changes.

tarvip, 2020-09-01:
Amlogic build seems to work, tried few games as well.
Amlogic-ng build fails in the end:
```
Making modifications to setres.sh...(ok)
Making modifications to emuelecRunEmu.sh...(ok)
Making modifications to BGM.sh...(ok)
Making modifications to advmame.sh...(ok)
Making modifications to es_settings.cfg...(ok)
Removing OpenBOR from es_systems.cfg./emuelec-addon.sh: line 1037: xmlstarlet: command not found
(failed)
```

tarvip, 2020-09-01:
Adding `xmlstarlet` locally helped.

tarvip, 2020-09-01:
Just to clarify, this pull request wont fix current master, I understand that development is happening in `dev` branches, eventually these changes will be moved to `master`.

shantigilbert, 2020-09-01:
Thank you!

Yes the changes will go to master once a new estable version is released!

tarvip, 2020-09-02:
@meeuuh92 if you pull latest changes from dev branch then you should be able to build it from `dev` branch: 
```
export GIT_BRANCH=dev
./emuelec-addon.sh
```

Also, before running `./emuelec-addon.sh` command comment out one row out [here](https://github.com/EmuELEC/EmuELEC/blob/a2449a996443c3c60975332bbe28bd6ee86e2c5b/emuelec-addon.sh#L1178-L1179), depending which version you need, saves time and disk space, at the moment both versions are built.

meeuuh92, 2020-09-02:
Thank you @tarvip... I've just started a compilation from dev branch.

meeuuh92, 2020-09-03:
I've restarted a new build of the dev branch last night... And after a lot of time... Everything seems ok.. (I 've not tested the addon yet  ; -)).. So with the changes you can compile emuelec on debian buster too

PR #285 (2020-09-01): EmuELEC addon build fixes
Fixes #228 

Maybe this README file update should be different as I am note sure that `xmlstarlet` package is needed for regular builds. 

After those changes I was able to build emuelec-addon with `emuelec-addon.sh` command only.
commit a2449a99 (2020-09-01): EmuELEC addon build fixes (#285)

* emulationstation-addon: add gcc target dependency

gcc package provides libgomp library

* mupen64plus-nx: fix Amlogic builds

PKG_URL was set before if block, it downloaded wrong version

* commander-genius: add Python3 target dependency

fixes following error:
error: 'PyUnicode_AsUTF8' was not declared in this scope

* README: add build dependency for addon build

xmlstarlet is needed for emuelec-addon.sh build

Fixes #228
closed 2020-09-03 by meeuuh92