EmuELEC/EmuELEC #229: Emulator options is missing

ws02589111, 2020-07-16:
I found the emulator options is missing on 3.6(build by myself), but 3.5 is ok(offical build), maybe I should add some options to emuelec.conf(default conf file now)?

here is path of the option
press select key(options) ->ADVANCED GAME OPTIONS-> EMULATOR(missing)
press select key(options) ->ADVANCED SYSTEM OPTIONS-> EMULATOR(missing)

TheLastProject, 2020-07-22:
What system is selected when you go to this option? My initial guess would be that there used to be multiple emulators for this system in 3.5 but not in 3.6.

ws02589111, 2020-07-22:
I tried fbn, nes, arcade and mame, but all of them missing too.
And then i checked the getcores.sh file ,the same with 3.5.
I tried to found some clue from the file
emuelec-emulationstation/es-app/src/guis/GuiMenu.cpp
but don't get any solution too
i guess in function popSpecificConfigurationGui， the systemData's file may empty?’

shantigilbert, 2020-07-22:
First of all since this is a version you compiled on your own, I need to know what branch did you use, if it was dev and If you upgraded from v3.5 you NEED to update binaries and scripts in the danger zone inside the emuelec settings menu.

getcores.sh does not exist anymore.

ws02589111, 2020-07-23:
I compiled the master branch and only update boot logos in \distributions\EmuELEC\splash.
After flash my build to card,  i copy my roms and bios from my old 3.5 system to the new system

shantigilbert, 2020-07-23:
Then sorry I have no idea, this is something only on your end as the emulator options are working fine on the 3.5, 3.6 and 3.7 test release

er0l, 2020-07-25:
Options for Amiga emulator are missing, while amiga cd32 games are shown in the folder of Amiga and if you enter that one you can see emulator options. This I've observed on all 3.7 test releases.

shantigilbert, 2020-07-25:
Open es_systems.cfg and delete all the <group> lines, restart emulationstation and options should be back

ws02589111, 2020-08-04:
> Then sorry I have no idea, this is something only on your end as the emulator options are working fine on the 3.5, 3.6 and 3.7 test release

thanks your reply, I updated to the latest dev branch, the issue is resolved

closed 2020-08-04 by ws02589111