EmuELEC/EmuELEC #230: Ports/Diablo fails to load

25minutes, 2020-07-22:
**Describe the bug**
I have copied diabdat.mpq to /storage/roms/ports/diablo/diabdat.mpq, but Diablo fails to load with error message "SDL Error: Audio device hasn't been opened. "

**To Reproduce**
Steps to reproduce the behavior:
1. Copy diabdat.mpq to /storage/roms/ports/diablo/.
2. In ES, try to launch Diablo under Ports section.
3. See error "SDL Error: Audio device hasn't been opened. "

**Expected behavior**
Per https://github.com/diasurgical/devilutionX, Diablo should load after diabdat.mpq is copied. Are there any additional steps needed to make it work under EmuELEC?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
Tested on both EmuELEC v3.5 & v3.6. Same results.

**Device (please complete the following information):**
S905m

**Additional context**
Add any other context about the problem here.

TheLastProject, 2020-08-15:
AJGJ reported the following on Discord

@shantigilbert Diablo 1 (Ports) problem solved !!
Today I woke up to an inspired day and along with a tip provided by a colleague at Discord of the devilution project. I decided to look for the files and the problem was in the Diablo.sh file. The audio drive in ALSA gave this error, I changed it to DEFAULT and the problem is solved. Game running and with 100% audio. Follow the corrected file in case anyone shows up here and has the same problem.
File location (emuelec / Configfiles / emuelec / ports) 

https://cdn.discordapp.com/attachments/570131631713550337/744167415507714048/picture01.png

@shantigilbert Could this help?

25minutes, 2020-08-16:
Indeed, it works! Thank you for sharing the info.

Here's the change to Diablo.sh, which works for me on S905m. I guess "default" doesn't work for me because I have set ee_alsa.always=1.

# init_port binary audio(alsa. pulseaudio, default)
init_port ${PORT} pulseaudio

TheLastProject, 2020-08-16:
Glad it works.

Could you please reopen the issue though? I want shanti to fix this for all users in the code :)

closed 2020-10-08 by shantigilbert