EmuELEC/EmuELEC #233: Ports/REminiscence (FLASHBACK) fails to load

25minutes, 2020-07-24:
**Describe the bug**
Try to run Ports/REminiscence (FLASHBACK), but it only shows blank screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Download data file from https://github.com/Cpasjuste/REminiscence/releases/download/v1.0/flashback_data.zip.
2. Copy "DATA" folder from "flashback_data.zip" to /storage/roms/ports/reminiscence
3. Try to run Ports/REminiscence (FLASHBACK), but it only shows blank screen.

**Expected behavior**
Flashback should load with data file being copied. The above steps are all that I can find from Internet, please give comments if additional steps are required to make it work on EmuELEC.

**Screenshots**
none.

**EmuELEC version:**
Only tested on EmuELEC v3.5.

**Device (please complete the following information):**
S905m

**Additional context**
EmuELEC Run Log
PLATFORM: ports
ROM NAME: /storage/roms/ports/reminiscence/
BASE ROM NAME: 
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/ports/reminiscence/
2nd Argument: -Pports
3rd Argument: --controllers=-p1index 0 -p1guid 050000004c050000cc09000000010000 
4th Argument: -Creminiscence
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence/
Emulator Output is:
[INFO] RetroArch 1.8.5 (Git 761a17ccd8)
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Apr  1 2020
[INFO] Version: 1.8.5
[INFO] Git: 761a17ccd8
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/reminiscence_libretro.so"
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 2.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [RCHEEVOS]: ext_hash 00001505 ('')
[INFO] Skipping SRAM load..
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [RCHEEVOS]: Load task finished
[INFO] [Cheats]: Load game-specific cheatfile: /tmp/database/cht/REminiscence/.cht
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00 Hz, Game = 50.00 Hz)
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [GL]: Found GL context: mali-fbdev
[INFO] [GL]: Detecting screen resolution 0x0.
[INFO] [EGL]: Current context: 0x40000001.
[INFO] [GL]: Vendor: ARM, Renderer: Mali-450 MP.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event3).
[INFO] [udev]: Keyboard #1 (/dev/input/mouse0).
[INFO] [udev]: Mouse #0 (/dev/input/event3).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: Wireless Controller (1356:2508) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 512 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[CDROM] No sg devices found and sg kernel module is not loaded.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x4eb78c
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[WARN] Input device ID 513 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].

shantigilbert, 2020-07-24:
Disable any shaders, flashback does not like them and if loaded with one will only show a black screen

25minutes, 2020-07-24:
Thanks shantigilbert. I activate RA quick menu and it says Video Shaders is OFF. Anything else could be the problem?

BTW, per https://docs.libretro.com/library/reminiscence/, the core accepts .map file. For troubleshooting purpose, I also try to manually load REminiscence (FLASHBACK) core from RA, then Load Content and manually specify LEVEL1.MAP from the DATA folder. Unfortunately, it still shows the black screen. I'm out of idea.

25minutes, 2020-07-25:
Here's an update. I still have a device running the old Sx05RE v2.2.1. I notice it also contains the REminiscence (FLASHBACK) core. When I manually launch LEVEL1.MAP, it indeed works. 

I think the problem might be related with the core version, so I export the core from Sx05RE and import it to EmuELEC v3.5 (storage/cores). Unfortunately it only shows black screen on EmuELEC v3.5 even if I'm using the old core.

shantigilbert, 2020-07-26:
For some reason this core does not like to be started by command line, so to load it you need to run it, then at the black screen open up the Retroarch menu, "close content" then on history tab reload the core and it should work.

shantigilbert, 2020-07-26:
Nevermind figure it out

on `REminiscence (FLASHBACK).sh` remote the leading / from the path so that it reads
`/emuelec/scripts/emuelecRunEmu.sh "/storage/roms/ports/reminiscence" -Pports "${2}" -Creminiscence "-SC${0}"`

shantigilbert, 2020-07-26:
fixed on https://github.com/EmuELEC/EmuELEC/commit/56fdea53f7eedcf1d61038e9ed5a6801a3c03857

25minutes, 2020-07-26:
Thanks shantigilbert for taking time on the analysis. It's strange that it still only shows black screen at my end. From the log file, I can see ROM NAME & 1st Argument are both "/storage/roms/ports/reminiscence" as you suggested.

What else am I missing? Just to double check, are you using the same data file as mine (https://github.com/Cpasjuste/REminiscence/releases/download/v1.0/flashback_data.zip)?

EmuELEC Run Log
PLATFORM: ports
ROM NAME: /storage/roms/ports/reminiscence
BASE ROM NAME: reminiscence
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/ports/reminiscence
2nd Argument: -Pports
3rd Argument: --controllers=-p1index 0 -p1guid 050000004c050000cc09000000010000 
4th Argument: -Creminiscence
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence
Emulator Output is:
[INFO] RetroArch 1.8.5 (Git 761a17ccd8)
[INFO] Redirecting save file to "/storage/savefiles/reminiscence.srm".
[INFO] Redirecting savestate to "/storage/savestates/reminiscence.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Apr  1 2020
[INFO] Version: 1.8.5
[INFO] Git: 761a17ccd8
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/reminiscence_libretro.so"
[INFO] [Overrides] no core-specific overrides found at /storage/.config/retroarch/config/REminiscence/REminiscence.cfg.
[INFO] [Overrides] no content-dir-specific overrides found at /storage/.config/retroarch/config/REminiscence/ports.cfg.
[INFO] [Overrides] no game-specific overrides found at /storage/.config/retroarch/config/REminiscence/reminiscence.cfg.
[INFO] [Remaps]: remap directory: /storage/remappings
[INFO] Redirecting save file to "/storage/savefiles/reminiscence.srm".
[INFO] Redirecting savestate to "/storage/savestates/reminiscence.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 2.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [RCHEEVOS]: ext_hash 00001505 ('')
[INFO] Skipping SRAM load..
[INFO] [RCHEEVOS]: Load task finished
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Cheats]: Load game-specific cheatfile: /tmp/database/cht/REminiscence/reminiscence.cht
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00 Hz, Game = 50.00 Hz)
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [GL]: Found GL context: mali-fbdev
[INFO] [GL]: Detecting screen resolution 0x0.
[INFO] [EGL]: Current context: 0x40000001.
[INFO] [GL]: Vendor: ARM, Renderer: Mali-450 MP.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event3).
[INFO] [udev]: Keyboard #1 (/dev/input/mouse0).
[INFO] [udev]: Mouse #0 (/dev/input/event3).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: Wireless Controller (1356:2508) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 512 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[CDROM] No sg devices found and sg kernel module is not loaded.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x4eb78c
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[WARN] Input device ID 513 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].

shantigilbert, 2020-07-26:
I will reopen this, but I tested it on 2 devices and it worked. :( don't know what is going on. I can't remeber where I got my dat file
I think mine came from http://cyxdown.free.fr/reminiscence/

shantigilbert, 2020-07-26:
Ok I used the data you have and it does load for me, but the "missing bios" on the ES menu will complains about missing level1.map the difference I see between your data files and mine are that the files are UPPERCASE on yours and all lowercase on mine.

Try changing the file level1.map to lowercase.

also since this is working fine for me, can you try a clean install? just to make sure there is no other issues with configuration?

25minutes, 2020-07-27:
Sure. I will test it today and let you know the result.

25minutes, 2020-07-27:
I make a new EmuELEC v3.6 image (EmuELEC-Amlogic.arm-3.6-Generic.img.gz on S905m). Unfortunately, it still shows black screen only. 

Here's what I have done:
1. Rename all data file names to lowercase.
2. Via "missing bios" ES menu, it appears to be looking for level1.map in /storage/roms/ports/reminiscence folder, instead of  ports/reminiscence/data or ports/reminiscence/DATA.
3. Copy all data files to both /storage/roms/ports/reminiscence & /storage/roms/ports/reminiscence/data. Just in case it looks for DATA folder, I also make a symlink (ln -s data DATA). So, no matter which path the core uses, it should be able to find the data files.
4. Verify "missing bios" no longer complain missing level1.map.
5. Edit /emuelec/ports/REminiscence (FLASHBACK).sh per your comment.
6. Restart ES.
7. Try to load the game, it only shows black screen.
8. At the black screen open up the Retroarch menu, "close content" then on history tab reload the core. It still shows black screen only.
9. Below is the EmuELEC Run Log for your reference.

EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 3.6 (Amlogic.arm)
PLATFORM: ports
ROM NAME: /storage/roms/ports/reminiscence
BASE ROM NAME: reminiscence
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/ports/reminiscence
2nd Argument: -Pports
3rd Argument: --controllers=-p1index 0 -p1guid 030000005e040000a102000000010000 
4th Argument: -Creminiscence
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence
Emulator Output is:
[INFO] RetroArch 1.8.8 (Git 439aee616e)
[INFO] [Overrides]: Redirecting save file to "/storage/.config/retroarch/saves/reminiscence.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/ports/reminiscence.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Jun 14 2020
[INFO] Version: 1.8.8
[INFO] Git: 439aee616e
[INFO] =================================================
[INFO] [CORE]: Loading dynamic libretro core from: "/tmp/cores/reminiscence_libretro.so"
[INFO] [Overrides]: No core-specific overrides found at /storage/.config/retroarch/config/REminiscence/REminiscence.cfg.
[INFO] [Overrides]: No content-dir-specific overrides found at /storage/.config/retroarch/config/REminiscence/ports.cfg.
[INFO] [Overrides]: No game-specific overrides found at /storage/.config/retroarch/config/REminiscence/reminiscence.cfg.
[INFO] [Remaps]: remap directory: /storage/remappings
[INFO] [Overrides]: Redirecting save file to "/storage/.config/retroarch/saves/reminiscence.srm".
[INFO] [Overrides]: Redirecting save state to "/storage/roms/ports/reminiscence.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 2.
[INFO] [CONTENT LOAD]: Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [SRAM]: Skipping SRAM load..
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Cheats]: Load game-specific cheatfile: /tmp/database/cht/REminiscence/reminiscence.cht
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00 Hz, Game = 50.00 Hz)
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [GL]: Found GL context: mali-fbdev
[INFO] [GL]: Detecting screen resolution 0x0.
[INFO] [EGL]: Current context: 0x40000001.
[INFO] [GL]: Vendor: ARM, Renderer: Mali-450 MP.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[... 196 lines cut ...]
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 512 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x540ae8
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].

shantigilbert, 2020-07-27:
I just tried it on another device (Odroid go Advance), and the black screen appeared, but I could load the game if I unloaded the content and started from the history, so I am very confused on how this core actually works, it works fine on my other 2 devices, it kinda works on the OGA and it does not work at all for you. :/ there is not much documentation.

25minutes, 2020-07-27:
Agree. I can hardly find much useful info about this core on Internet. I'm not really a fan of this game. As an engineer (though I'm not coding), I'm just curious to understand how things work. :)

Please feel free to close the ticket for now. We can revisit it if you come up with some new idea later and need someone to test it.

shantigilbert, 2020-07-29:
Can you try and run this core from the CLI ? 
The data should be unzipped directly into  `/storage/roms/ports/reminiscence`  (no `DATA` folder) then ssh to your device and type this

```
systemctl stop emustation
/usr/bin/retroarch --verbose -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence
``` 

I am trying to figure out if this is an issue with ES or something else.

25minutes, 2020-07-30:
Good news. It works properly with this method!

Comparing the command line from the previous EmuELEC Run Log, the only difference is, it was using "-v" instead of "--verbose". Then I also try the following, but it still works fine.

systemctl stop emustation
/usr/bin/retroarch -v -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence

So the magic seems to be "systemctl stop emustation"?  Below are the output at the console for your reference.

EmuELEC:~/roms/ports/reminiscence # /usr/bin/retroarch --verbose -L /tmp/cores/reminiscence_libretro.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/ports/reminiscence
[INFO] RetroArch 1.8.5 (Git 761a17ccd8)
[INFO] Redirecting save file to "/storage/savefiles/reminiscence.srm".
[INFO] Redirecting savestate to "/storage/savestates/reminiscence.state".
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4 ASIMD
[INFO] Built: Apr  1 2020
[INFO] Version: 1.8.5
[INFO] Git: 761a17ccd8
[INFO] =================================================
[INFO] Loading dynamic libretro core from: "/tmp/cores/reminiscence_libretro.so"
[INFO] [Overrides] no core-specific overrides found at /storage/.config/retroarch/config/REminiscence/REminiscence.cfg.
[INFO] [Overrides] no content-dir-specific overrides found at /storage/.config/retroarch/config/REminiscence/ports.cfg.
[INFO] [Overrides] no game-specific overrides found at /storage/.config/retroarch/config/REminiscence/reminiscence.cfg.
[INFO] [Remaps]: remap directory: /storage/remappings
[INFO] Redirecting save file to "/storage/savefiles/reminiscence.srm".
[INFO] Redirecting savestate to "/storage/savestates/reminiscence.state".
[INFO] [Environ]: GET_LOG_INTERFACE.
[INFO] [Environ]: PERFORMANCE_LEVEL: 2.
[INFO] Content loading skipped. Implementation will load it on its own.
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[INFO] [Environ]: SET_PIXEL_FORMAT: XRGB8888.
[INFO] [RCHEEVOS]: ext_hash 00001505 ('')
[INFO] Skipping SRAM load..
[INFO] [RCHEEVOS]: Load task finished
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Cheats]: Load game-specific cheatfile: /tmp/database/cht/REminiscence/reminiscence.cht
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 60.00 Hz, Game = 50.00 Hz)
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [GL]: Found GL context: mali-fbdev
[INFO] [GL]: Detecting screen resolution 0x0.
[INFO] [EGL]: Current context: 0x40000001.
[INFO] [GL]: Vendor: ARM, Renderer: Mali-450 MP.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [udev]: Keyboard #0 (/dev/input/event3).
[INFO] [udev]: Keyboard #1 (/dev/input/mouse0).
[INFO] [udev]: Mouse #0 (/dev/input/event3).
[INFO] [udev]: Mouse #1 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: Wireless Controller (1356:2508) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event4) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 512 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[CDROM] No sg devices found and sg kernel module is not loaded.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x4eb78c
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[WARN] Input device ID 513 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [config] Saved new config to "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Cheats]: Save game-specific cheatfile: /tmp/database/cht/REminiscence/reminiscence.cht
[INFO] Content ran for a total of: 00 hours, 00 minutes, 34 seconds.
[INFO] Unloading game..
[INFO] Unloading core..
[INFO] Unloading core symbols..
[INFO] Threaded video stats: Frames pushed: 1723, Frames dropped: 0.

shantigilbert, 2020-07-30:
systemctl stop emustation is to kill emulationstation (ES) 
-v is just short for --verbose so it makes no difference, but the problem might be that the core does not like being launched by ES, I will have to do more investigating and figure out why it works on some devices and not on others.

25minutes, 2020-07-30:
Okey. Thanks shantigilbert!

25minutes, 2020-07-30:
A quick question. I try to edit "REminiscence (FLASHBACK).sh" (see below) as a temp workaround, but it doesn't work (only shows black screen). Any idea?

#!/bin/bash

systemctl stop emustation
/emuelec/scripts/emuelecRunEmu.sh "/storage/roms/ports/reminiscence" -Pports "${2}" -Creminiscence "-SC${0}"
systemctl start emustation

shantigilbert, 2020-07-30:
You're killing es which is what runs the script you just called.

25minutes, 2020-07-30:
I see. Thanks again.

shantigilbert, 2020-08-05:
So have we determined that this is a problem when launching it from ES only?

25minutes, 2020-08-06:
Yes, it is. Anything else you want me to test?

miwasp, 2021-02-11:
The problem still exists in version 4.0. The workaround

> open up the Retroarch menu, "close content" then on history tab reload the core and it should work.

still works in 4.0.

shantigilbert, 2021-06-07:
This has been fixed, at least I can no longer reproduce on v4.2-test. If its not fixed please reopen the issue.

closed 2021-06-07 by shantigilbert

dre10g, 2021-07-15:
This is still not working in 4.2 on Gameforce Chi
The workaround still works though - open up the Retroarch menu, "close content" then on history tab reload the core and it should work.

schmurtzm, 2022-11-08:
Hi, it could help :

Take a look here : [Blackscreen on many platforms when running by command line](https://github.com/libretro/REminiscence/issues/13) 

If you can make a try with a retroarch.cfg without the lines which start with `input_player` (may be keep lines which start with `input_player1_`) and make a feedback please ?

If you have `Settings` -> `Configuration` -> `Save Configuration on Quit` enabled in Retroarch settings it will re-create these entries automatically when you exit Retroarch and break REminiscence again so I recommend to create a dedicated config file (with this option disabled) that you load with `--config` in your command line 😉

commit 56fdea53 (2020-07-26): REminiscence: Fix  black screen, fixes #233