EmuELEC/EmuELEC #241: Change to other Font for ozone failed on EE3.6

yingw, 2020-07-30:
Before 3.6 I used to change some Chinese font for RetroArch ozone theme, the steps:
1. in ftp, transfer **regular.ttf**, and **bold.ttf** to `/tmp/assets/ozone`
2. Exit to RetroArch, select Chinese in `User` > `Language` menu
3. Done. Everytime restarted, the fonts remains.

But in 3.6, I did the exactly steps as before, but It just show normal once when I change in menu. After restart RA, all menu words turns into ????, looks like an default EN fonts. If I switch to English in menu, it shows en normally, but is some Pixel Font (Strange!) not the default one.

And check the `/tmp/assets` and `/storage/assets/ozone`, the Chinese fonts remains there.

I tested several times. And tested in EE 3.4, 3.5, everything works fine. Looks like a bug in 3.6.

BTW, I noticed, In 3.4, 3.5, the menu just the language name, such as "English", the 3.6 shows "English (need restart...)", Is there any extra step to "restart"?

yingw, 2020-07-30:
The normal default fonts:

[image]

After Updated CN fonts:
[image]

[image]
and 
[image]

After restart:
[image]
and (What's this Font?)
[image]

shantigilbert, 2020-08-02:
Can you test v3.7-TEST ? https://github.com/EmuELEC/EmuELEC-tests/releases

yingw, 2020-08-03:
> Can you test v3.7-TEST ? https://github.com/EmuELEC/EmuELEC-tests/releases

Hi, Tested. But unfortunately it failed the same as 3.6.

After reenter RetroArch, it shows ???? again.

Change to EN, the strange Font:

[image]

Save config file, reenter RetroAerch, it return to normal Font.

[image]

shantigilbert, 2020-08-03:
Sorry then, I have no idea why that happens as I have never used changed fonts. :(

shantigilbert, 2020-11-26:
This should be fixed now, please reopen the issue if its not.

closed 2020-11-26 by shantigilbert

yingw, 2020-11-26:
> This should be fixed now, please reopen the issue if its not.

Thanks, but I had fixed it by copying the 'pkg' folder from <https://github.com/libretro/retroarch-assets> to the '/tmp/assets' folder. And this is also mentioned and solved in simular solution by other guys using Chinese.

I will test the new version sometime.

shantigilbert, 2020-11-26:
That is the same "fix" I applied.

yingw, 2021-07-29:
@shantigilbert hi, found it breaks ag. in 4.1 and 4.2, and lot report from some Chinese forum：https://tieba.baidu.com/p/7319507359

[image]

just change to Chinese (whether change fort in /tmp/assets/ozone or not) , save config, restart, it comes 口口口( not the ??? last time), if using English as default lang and change to Chinese after entering RA, it works normally.

found you have updated the pkg folder in the last fix, and I tried to update this time, not working.

shantigilbert, 2021-07-29:
https://github.com/EmuELEC/EmuELEC/issues/534

commit 0c81cab8 (2020-06-04): squashfs-tools: fix build failure against gcc-10
This close #241.