EmuELEC/EmuELEC #242: Dev and master branch build failed

ws02589111, 2020-07-31:
I have tested master and dev branch
The following package install get 404：
https://sources.libreelec.tv/mirror/portaudio/portaudio-190600_20161030.tgz
and then the script tried other download link：
 http://www.portaudio.com/archives/pa_stable_v190600_20161030.tgz
is also get the error：
Read error (Connection timed out) in headers

FAILURE: scripts/install portaudio has failed!
[386/423] [FAIL] install portaudio

shantigilbert, 2020-07-31:
seems like the portaudio site is down, not much I can do about that, either find an alternative or wait until its restored.

shantigilbert, 2020-07-31:
Check https://github.com/EmuELEC/EmuELEC/commit/7b3e25e02a5b12e4769081bdc4404e4ea0107c1b

closed 2020-07-31 by shantigilbert

ws02589111, 2020-07-31:
The mupen64plus-nx package also the problem

shantigilbert, 2020-07-31:
no it does not, https://github.com/libretro/mupen64plus-libretro-nx/commit/be9b85ad17c7bcad5de353ad22e81c7bf6d77c48
check your internet

shantigilbert, 2020-07-31:
well just change it to whatever checksum you got

commit b2d2c16a (2020-07-31): Portaudio: Use github

Since portauido.com seems to be down 
Closes #242
commit 7b3e25e0 (2020-07-31): Portaudio: Use github

Since portaudio.com seems to be down 
Closes #242