EmuELEC/EmuELEC #249: Emuelec 3.6 C16/Plus 4 system issue

Antras75, 2020-08-05:
I put several roms in .prg format in the C16 folder, I refreshed the game list, but the system does not appear in the main menu. I use Emuelec 3.6 with GT KING Amlogic S922X

shantigilbert, 2020-08-05:
Edit the es_systems.cfg and add it to the extensions lists

```
<extension>.prg .PRG .d64 .g64 .t64 .x64 .zip .D64 .G64 .T64 .X64 .ZIP</extension>
```

Antras75, 2020-08-05:
Thanks Shantigilbert, now "Commodore 16" system appear in system menù, but not load core... When i launch a game nothing happens... Only return to Retroarch menù with in left corner 1.8.8 - No core. In the cores list on Retroarch VICE_XPLUS4_LIBRETRO are present... I not understand.

shantigilbert, 2020-08-05:
post the complete emuelec.log right after you try to launch a game

also make sure the ocre is actuall installed in /tmp/cores

Antras75, 2020-08-06:
Yes, i check TMP/CORES folder, and vice_xplus4_libretro.so are present...

shantigilbert, 2020-08-06:
"post the complete emuelec.log right after you try to launch a game"

Antras75, 2020-08-06:
EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 3.6 (Amlogic-ng.arm)
PLATFORM: c16
ROM NAME: /storage/roms/c16/Fire Ant.prg
BASE ROM NAME: Fire Ant.prg
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/c16/Fire Ant.prg
2nd Argument: -Pc16
3rd Argument: --controllers=
4th Argument: 
Run Command is:
/usr/bin/retroarch -v -L /tmp/cores/vice_xplus4_libretro.so --set-shader /tmp/shaders/rpi/crt-pi-curvature.glslp --config /storage/.config/retroarch/retroarch.cfg /storage/roms/c16/Fire Ant.prg
Emulator Output is:
[INFO] RetroArch 1.8.8 (Git 439aee616e)
[INFO] [Overrides]: Reindirizzare il file di salvataggio su "/storage/.config/retroarch/saves/Fire Ant.srm".
[INFO] [Overrides]: Reindirizzare il salva stato su "/storage/roms/c16/Fire Ant.state".
[INFO] === Build =======================================
[INFO] Funzionalità:  NEON ASIMD
[INFO] Built: Jun 14 2020
[INFO] Version: 1.8.8
[INFO] Git: 439aee616e
[INFO] =================================================
[INFO] [CORE]: Loading dynamic libretro core from: "/tmp/cores/vice_xplus4_libretro.so"
[ERROR] Impossibile aprire il core libretro: "/tmp/cores/vice_xplus4_libretro.so"
Error(s): /tmp/cores/vice_xplus4_libretro.so: undefined symbol: c64memrom_kernal64_rom
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Overrides]: Reindirizzare il file di salvataggio su "/storage/.config/retroarch/saves/Fire Ant.srm".
[INFO] [Overrides]: Reindirizzare il salva stato su "/storage/roms/c16/Fire Ant.state".
[INFO] Versione delle API di libretro: 1
[INFO] Compilate con API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [GL]: Found GL context: mali-fbdev
[INFO] [GL]: Detecting screen resolution 0x0.
[INFO] [EGL]: Current context: 0xf2a44568.
[INFO] [GL]: Vendor: ARM, Renderer: Mali-G52.
[INFO] [GL]: Version: OpenGL ES 3.2 git.db77a4c.74502ed16b0fc2da36c811f236840b85.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 1114 frames
[INFO] ALSA: Buffer size: 2229 frames
[INFO] [Display]: Found display driver: "gl".
[ERROR] Impossibile aprire il core libretro: "/tmp/cores/vice_xplus4_libretro.so"
[ERROR] Error(s): /tmp/cores/vice_xplus4_libretro.so: undefined symbol: c64memrom_kernal64_rom
[INFO] [Display]: Found display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x540b3c
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[INFO] [SRAM]: La SRAM non verrà salvata.
[INFO] [Playlist]: Caricamento dei file della cronologia: [/storage/.config/retroarch/content_history.lpl].
[INFO] [Playlist]: Caricamento dei file della cronologia: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] [Playlist]: Caricamento dei file della cronologia: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] [Playlist]: Caricamento dei file della cronologia: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Playlist]: Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [config] Salvata la nuova configurazione su "/storage/.config/retroarch/retroarch.cfg".
[INFO] Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [CORE]: Unloading core..
[INFO] [CORE]: Unloading core symbols..
[?12l[?25h[INFO] Threaded video stats: Frames pushed: 149, Frames dropped: 0.
The framebuffer device was opened successfully.
vinfo.yoffset=1080
FIX: setting vinfo.yoffset=0.

shantigilbert, 2020-08-06:
seems there might be an issue with the core 

```
[ERROR] Error(s): /tmp/cores/vice_xplus4_libretro.so: undefined symbol: c64memrom_kernal64_rom
```

Antras75, 2020-08-06:
Yes, i'm read too.... Now i try to replace the core maybe are corrupted or something...

Antras75, 2020-08-06:
I try to replace Vice_xplus4_libretro.so and still don't work... But now give me this error log:
[ERROR] Error(s): /tmp/cores/vice_xplus4_libretro.so: wrong ELF Class: ELFCLASS64

shantigilbert, 2021-02-10:
This should be fixed on v4.0 if its not feel free to reopen the issue.

closed 2021-02-10 by shantigilbert