EmuELEC/EmuELEC #250: Report v3.7 stable on OGA

zombiekutez, 2020-08-07:
Model: OGA v1.
Using: Emuelec v3.7 80e6468
Fresh install.
1. Drop performance on 3d emulator. Mupen64plus_libretro from 19~21 fps on v3.6 => 13~15fps on v3.7
Flycast_libretro from ~ 20fps on v3.6 => ~15 fps on v3.7
PPSSPPSDL from ~16fps on v3.6 => ~10 fps on v3.7
2. Pcsx_rearmed crash from boot in all game. I using exactly bios run on v3.6.
3. New reicast core add not running, crash from boot. And why we can't using Redream like Retropie and orther for now ???

Some good side of v3.7 i can see here. Verry better menu UI, better graphics on 2d emulator like Sega cd, Snes, nice color, better smoother, sharper and definition. 
Thank. Hope you have a nice day.

shantigilbert, 2020-08-07:
1.- Did you try a clean build or upgrade? if upgrade did you upgrade correctly ? and can you try a clean build to make sure its not settings related. 
2.- logs!
3.- Ask the Redream team.

zombiekutez, 2020-08-07:
> 1.- Did you try a clean build or upgrade? if upgrade did you upgrade correctly ? and can you try a clean build to make sure its not settings related.
> 2.- logs!
> 3.- Ask the Redream team. (not sure what you mean by new reicast core, Reicast is not included on OGA)

I get install v3.7 clean on a new sd card, but i think low performance may cause my new sd card it is 32gb Class 4, my v3.6 run on 512gb class 10 A2 U1.
You can test your OGA on v3.7, any game are crash, if you don't crash maybe cause i got some wrong thing.
In Dreamcast selection acturely show a Reicast core ??? I don't know how, but sorry.

TheLastProject, 2020-08-08:
I'm not having any game crashes on 3.7. Please start the game and after the crash go to `Setup / Scripts -> 08 - send logs` and put the link here.

zombiekutez, 2020-08-09:
Sorry i only don't know why but i start i fresh install and choose reset system to defaut config and it booted.

closed 2020-08-09 by zombiekutez