EmuELEC/EmuELEC #253: v3.7 error in AdvanceMame when exiting any game

miwasp, 2020-08-07:
Upgrade from v3.6. Romset for AdvanceMame in folder "arcade". An error is raised every time when exiting a game (1+6 twice). This worked in v3.6. Sample error log:

EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 3.7 (OdroidGoAdvance.arm)
PLATFORM: arcade
ROM NAME: /storage/roms/arcade/1942.zip
BASE ROM NAME: 1942.zip
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/arcade/1942.zip
2nd Argument: -Parcade
3rd Argument: --core=AdvanceMame
4th Argument: --emulator=AdvanceMame
Full arguments: /storage/roms/arcade/1942.zip -Parcade --core=AdvanceMame --emulator=AdvanceMame --controllers=-p1index 0 -p1guid 19000000010000000200000011000000 
Run Command is:
/usr/bin/bash /usr/bin/advmame.sh /storage/roms/arcade/1942.zip
Emulator Output is:
librga:RGA_GET_VERSION:4.00,4.000000 
ctx=0x30db058,ctx->rgaFd=7 
Rga built version:version:+2017-09-28 10:12:42 
rk-debug mali-ver=rsx-p0-1.0
ALSA lib /mnt/Sx05RE_SSD/CoreELEC/build.EmuELEC-OdroidGoAdvance.arm-3.7/alsa-lib-1.1.9/src/pcm/pcm.c:2564:(snd_pcm_open_noupdate) Unknown PCM sdl
Double signals, 15 and 11
Signal SIGSEGV[MAPERR], fault at 0xf0422b61, from code at (nil)
Compiled Aug  6 2020, 03:43:50
exit 1

mymo1982, 2020-08-14:
I am experiencing the same. Also, unlike Emuelec 3.6, advancemame doesn't keep per game configurations.

shantigilbert, 2020-08-14:
https://github.com/EmuELEC/EmuELEC/commit/982abe8318a630aaa5f877de271d86b79540cb57

closed 2020-10-08 by shantigilbert