EmuELEC/EmuELEC #256: [Bug] Unresponsive Saturn controls (default OGA controls only)

Cee123, 2020-08-08:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
On the Odroid Go Advance, there is a bug where the default Odroid Go Advance controls for Sega Saturn won't be responsive and the controls will not detect within Retroarch. BUT if you plug in another controller or keyboard, it's perfectly playable.

**To Reproduce**
Steps to reproduce the behavior:
Controls not responding.

**Expected behavior**
Saturn runs perfectly fine and there are games that are perfectly playable on the Odroid Go Advance, but that bug with the Odroid Go Advance's controls not working is a bit of a pain.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
3.6 & 3.7

**Device (please complete the following information):**
Odroid Go Advance - the default controls. Using an alternative controller, and even a keyboard it is perfectly playable. The default OGA controls aren't detected like with every system.

**Additional context**
I've also run into the same issue on EmuElec when adding Saturn myself on version 3.6. The default Odroid Go controls work and are automatically picked up by every other system upon loading a game. But for some odd reason, this doesn't happen with Sega Saturn. If you plug in a USB controller or keyboard, then you can play it. And Saturn does have games that run perfectly well on the Odroid Go Advance. But the bug is highly unusual and prevents the default controls from detecting and working on retroarch.

shantigilbert, 2020-08-08:
This sounds more like a Core issue than an EmuELEC issue, Retroarch handles input, so if the input works on any other Retroarch core they should in theory work for the Saturn core, if they do not I assume its a core issue.

shantigilbert, 2020-08-09:
Already tried a lot of other older versions, with different arguments to compile and they all act the same, so I have no idea how to fix this, it seems to be in the core.

Cee123, 2020-08-09:
> 
> 
> This sounds more like a Core issue than an EmuELEC issue, Retroarch handles input, so if the input works on any other Retroarch core they should in theory work for the Saturn core, if they do not I assume its a core issue.

Yeah it's strange. At first I thought it might've been an issue with Retroarch. So I tried changing the drivers under the driver menu. But that didn't do anything either.

Interestingly, I also managed to add Saturn to the retroarena build using a 64 bit core I took from my Odroid N2. And it didn't have the controller issue. So maybe it could be something to do with the 32 Bit core.

I remember RetroArena had an issue with N64 controls on standalone Mupen64plus that they said was due to the kernel.

shantigilbert, 2020-08-10:
From which distro did you pulled that core? I tested it on EmuELEC 64bit and it seems to work fine

Cee123, 2020-08-10:
> From which distro did you pulled that core? I tested it on EmuELEC 64bit and it seems to work fine
> 
> Edit: scratch that, after a bit it started doing the same thing :/
> 
> edit 2: its still a LOT more playable, I will test for ARM and see if it fixes it.
> 
> Ok so this seems to be an issue with the 32bit version... this will have to wait until EE 4.0

Very interesting. Yeah it's a total mystery. Since i dont know how to compile cores, I copied the .so file from RetroArena for Odroid N2. I figured since it's 64 bit Arm it should also work on retroarena's 64 bit retroarch on the OGA. I decided to try it. Seemed to work ok. Then I also tried the 32 bit Arm file i took from RetroArena for the Odroid XU4 and that also had the issue.

SumavisionQ5, 2020-08-10:
> Very interesting. Yeah it's a total mystery. Since i dont know how to compile cores, I copied the .so file from RetroArena for Odroid N2. I figured since it's 64 bit Arm it should also work on [...]

Thanks for the info. I will try it later.

shantigilbert, 2020-10-08:
Fixed on 3.8

closed 2020-10-08 by shantigilbert

Cee123, 2020-10-10:
Good work guys.