EmuELEC/EmuELEC #257: Crash dump when exiting games

Cee123, 2020-08-08:
**Describe the bug**
Every time I exit a game on the Odroid Go Advance, version 3.7, I get a crash dump describing a bunch of things. Happens with every system.

**To Reproduce**
I recently updated from version 3.6. I did as described according to the Emuelec page, and started again with my es_systems.cfg, Emuelec options etc as well as resetting both the retroarch and emuelec scripts and binaries under the "Danger Zone" options.

**Expected behavior**
Should just exit the game like normal, like in the previous version.

**Screenshots**
[image]

**EmuELEC version:**
Version 3.7

**Device (please complete the following information):**
Odroid Go Advance

**Additional context**
I've been trying to read the whole thing, but it's difficult to read sideways. I try to scroll down, but before I can even get to half way of the long error message, it flashes off the screen and back to the Emulationstation. If you try to close it, it comes up again. So you must scroll down to read it and then it goes away before you've even finished reading it. I've tried searching the SD card for some kind of crash dump to see what it says, but no such luck.

TheLastProject, 2020-08-08:
Please go to `Scripts / Setup -> 08 - send logs` after the crash (it's in the home menu with all the other systems), it will upload the latest log and you can paste the URL here so we can look at the crash.

Cee123, 2020-08-08:
Ok, thanks for that! I'll try that.

Cee123, 2020-08-08:
Alright so I've resolved this issue now. I'm thinking a couple of the config files got messed up during the updating process. But I fixed it by resetting the emuelec scripts and binaries under the Danger Zone options after a second time. I did do this once, but I don't think it worked the first time.

closed 2020-08-08 by Cee123