EmuELEC/EmuELEC #258: BIOS prompts & can't change emulator for systems

Cee123, 2020-08-08:
**Describe the bug**
I keep getting prompts telling me I don't have certain bios files for systems. But these BIOS files aren't needed by the emulator. Why does Atari 7800 now suddenly need a bios file? Flycast doesn't use naomi_bios.bin. I have the Sega CD and Neo Geo bios files, but it keeps saying I don't have them, and then it goes into the game anyway. It requests bios files that I've already got in the /roms/bios folder. For example I do have naomi.zip in the /bios/dc folder and the Naomi games work, but the prompt always comes up.

Another issue I've noticed with the new version is you can't change the emulator for a particular system if there is only one emulator listed for it. So what I did was I go into es_systems.cfg and add a new <core>Test</core> for that system. And then in the Emuelec menu, I can now change the emulator for that system and it brings up the usual settings again. Otherwise it just says "Auto" now and displays one setting "Enable Max Performance". But when it's on Auto and it tries to launch a game there is no emulator in use for that system.

**To Reproduce**
-

**Expected behavior**
It was easier to change emulators in the previous version. Also was better without the prompts for bios files.

**Screenshots**

**EmuELEC version:**
3.7

**Device (please complete the following information):**
Odroid Go Advance

**Additional context**
Add any other context about the problem here.

TheLastProject, 2020-08-08:
The BIOS prompt is not a bug but a feature and I strongly disagree it was better without the prompt. It tells you it detects files are missing and the game MAY not work. We try anyway, so that a false positive doesn't keep you from being able to play, but it is to help beginners understand why their game won't start. If you can however tell us which files the prompt tells you you're missing but apparently aren't necessary we can tweak the list of files it should check for. We're using Batocera's list, but they never separated required and optional BIOS files.

On the changing the emulator list, did you use the RESET EMUELEC SCRIPTS AND BINARIES TO DEFAULT option yet?

Cee123, 2020-08-08:
But the prompts still come up even with the necessary bios files in the directory.

TheLastProject, 2020-08-08:
Again, which files?

Cee123, 2020-08-08:
It comes up with
naomi.zip and i have that in the roms\bios\dc directory 
bios_CD_U.bin and bios_CD_J.bin i have those in the roms\bios directory as well 
Also neogeo.zip which i also have in the roms\bios directory

shantigilbert, 2020-08-08:
> It comes up with
> naomi.zip and i have that in the roms\bios\dc directory
> bios_CD_U.bin and bios_CD_J.bin i have those in the roms\bios directory as well
> Also neogeo.zip which i also have in the roms\bios directory
> 
> I dont know unless they are case sensitive or something or i havent placed them in the right folders. Could be something very simple.

The neogeo bios needs to go in the neogeo folder (it says so right on the prompt!)

`"neogeo":   { "name": "NeoGeo", "biosFiles":    [ { "md5": "", "file": "neogeo/neogeo.zip" } ] },`

and since this is Linux, files ARE case sensitive

> But the prompts still come up even with the necessary bios files in the directory.

Not it does not, you need to pay attention to what is asked for (your files might be invalid not just missing), the whole reason the bios check is there, is to avoid stuff like this, but it won't help anyone if they don't pay attention to the error message. 

The bioscheck is not going to go away, I personally like this feature and I know it can help reduce issues like this one, we can tweak it if we missed some bios or if the files are not needed/wrong as it is common for new features like this to present issues since we cannot test with all the different variables that exists, but it will get better.

TheLastProject, 2020-08-08:
When I put a naomi.zip in `roms/bios/dc`, I no longer get a warning I need naomi.zip, but that my naomi.zip is `UNTESTED` (not missing) because the MD5sum isn't correct, and that I'm missing naomi_boot.bin in `roms/bios/dc`.

Please be very precise with the errors you get. Take a picture if you can. Then we can go and find out if there are any actual issues that need fixing.

shantigilbert, 2020-08-08:
as for this (you should've really opened 2 issues as they have nothing to do with each other!)

>Another issue I've noticed with the new version is you can't change the emulator for a particular system if there is only one emulator listed for it. So what I did was I go into es_systems.cfg and list a new Test core for that system. And then in the Emuelec menu, I can now change the emulator for that system and it brings up the usual settings again. Otherwise it just says "Auto" now and displays one setting "Enable Max Performance". But when it's on Auto and it tries to launch a game there is no emulator in use for that system.

Since you are upgrading, make sure you've read and followed the instructions posted on the release notes!

IF you are still getting issues even after a correct update, then you forgot to mention which system this happens on so I can look at it.

Cee123, 2020-08-09:
> The neogeo bios needs to go in the neogeo folder (it says so right on the prompt!)
>
> `"neogeo": { "name": "NeoGeo", "biosFiles": [ { "md5": "", "file": "neogeo/neogeo.zip" } ] },`
>
> and since this is [...]

Ohhh ok. It goes in the neogeo folder. Would that happen to be /roms/neogeo or /roms/bios/neogeo? I'll try both.

Yeah I figured they'd be case sensitive especially with Linux. I've made sure the cases match and everything.

Cee123, 2020-08-09:
> Not it does not, you need to pay attention to what is asked for (your files might be invalid not just missing), the whole reason the bios check is there, is to avoid stuff like this, but it won't help anyone if they don't pay attention to the error message.
> 

Interesting, so they are the wrong file versions. And that's what the whole UNTESTED thing means? But it's strange because they do work.

>When I put a naomi.zip in roms/bios/dc, I no longer get a warning I need naomi.zip, but that my naomi.zip is UNTESTED (not missing) because the MD5sum isn't correct, and that I'm missing naomi_boot.bin in roms/bios/dc.

So you're saying there are different versions of these BIOS files? I could be using the wrong versions with mismatched file sizes. I'll do some searching, and try and find some other versions and I'll try those.

Cee123, 2020-08-09:
This is the only option that came up when I tried to set mednafen_vb as the default emulator for Virtual Boy. But when I added i.e. Test as another core under Virtual Boy in es_systems.cfg I was able to change and select it.

[image]

Sorry about the terrible quality. Stupid phone camera.

You're right, I really should have made this a separate issue. Maybe I should repost it as separate issue. And thanks, I'll try those instructions again on the release notes. Although the issue seems to be resolved now with my little workaround.

shantigilbert, 2020-08-09:
I don't understand, why would it give yo an option to change emulator, if there is only one emulator? adding another emulator will obviously show you the option to change it, since there are 2, but if there is only one... Maybe I am not understanding what you are trying to accomplish here.

Cee123, 2020-08-10:
After I upgraded and I did the Danger Zone reset the emulationstation scripts and settings, for some strange reason there wasn't a default emulator set for each system. I'd go to load a game and it would crash dump. So I had to change it manually.

shantigilbert, 2020-08-10:
You should have probably started the issue with that. 

if that is actually y what happened then the es_systems.cfg did not get copied, and it should have which makes it another issue.

Cee123, 2020-08-10:
Ok guys, I've fixed the whole BIOS thing by getting the correct versions of the bios files, so nevermind about that. I just wasn't aware the BIOS files I was using were not the officially supported ones.

closed 2020-08-11 by shantigilbert

Cee123, 2020-08-11:
I've tried all the versions of the Naomi.zip bios I can find, and it is still coming up with the same thing - UNTESTED.

shantigilbert, 2020-08-11:
You are not looking hard enough, I literally just searched for the md5 and found a lot of links to the naomi.zip file.

Cee123, 2020-08-11:
> You are not looking hard enough, I literally just searched for the md5 and found a lot of links to the naomi.zip file.

Thanks. I'll try searching that checksum.

Cee123, 2020-08-11:
> 
> 
> You are not looking hard enough, I literally just searched for the md5 and found a lot of links to the naomi.zip file.

What is the checksum of the file? Just tried a bunch of different Naomi.zip files, they all work. So why does it need one with a specific checksum/md5?

Cee123, 2020-08-11:
https://wiki.batocera.org/doku.php?id=what_are_bios_and_how_to_add_them

According to the Batocera this is the checksum for the bios/naomi.zip eb4099aeb42ef089cfe94f8fe95e51f6 

I think there's something wrong here. I have got a BIOS file of naomi.zip with exactly that checksum, and I'm still getting the same error. I've searched everywhere all over the internet, downloaded a bunch of different naomi.zip files and I'm getting the same error with each one.

Cee123, 2020-08-11:
Ok, problem solved. After searching high and low I FINALLY found a BIOS file with the checksum
e63d892cdb8b532351dc7020bb60b6f4

So for anyone else having this frustrating issue, the BIOS file needs the specific checksum/md5 of -- e63d892cdb8b532351dc7020bb60b6f4
But all others will work, except they will show the prompt each time upon loading a game.

I tried a bunch of other naomi.zip files with the checksums of
219dee1350d9268125cdd3cec19d04c0
391f8dcf1b69481df2dd8c1e90aae1a9
6406289cbe38d7a7544d2345dde87028
961197561691dc66b1add26f668ba03d
be6b65cd065b41a1e82ac12dc4be2c30
eb4099aeb42ef089cfe94f8fe95e51f6

For anyone wondering what the checksum/MD5 is of a bios file, use the software WinMD5.

TheLastProject, 2020-08-11:
The UNTESTED screen actually shows what checksum is recommended. You can also see this info by going to the Menu, choosing "Games Settings" and choosing "Missing Bios".

Cee123, 2020-08-11:
> The UNTESTED screen actually shows what checksum is recommended. You can also see this info by going to the Menu, choosing "Games Settings" and choosing "Missing Bios".

You're right. I got confused thinking that was the checksum of the bios i was using hahaha and was wondering why it was giving me that same md5 with each bios i tried. Then i realized that was the checksum of the bios i needed. /facepalm Ah well at least i know now.