EmuELEC/EmuELEC #261: Wifi not work on x96 mini 2/16 (Amlogic S905W)

bennington111, 2020-08-09:
Try to wifi connect from Emuelec mine menu.
Try to wifi connect from menu Setup - Wifi.
Try to wifi connect from Retroarch.
Not work. 
Error /net/connman/service/wifi_....._managed_psk: Input/output ... connection failed.
Error /net/connman/service/wifi_......_managed_psk:Not connected ... disconnection failed.

[image]

cponzoni, 2020-08-18:
i confirm wifi is not working on x96 mini ( s905w )

shantigilbert, 2020-08-18:
You need to SSH to your device and type 

```
udevadm info /sys/bus/sdio/devices/sdio* | pastebinit
``
then paste the resuting url here

bennington111, 2020-08-18:
> You need to SSH to your device and type
> 
> ```
> udevadm info /sys/bus/sdio/devices/sdio* | pastebinit
> ```
> 
> then paste the resuting url here

http://ix.io/27P3

Alfa83, 2020-08-31:
Hola, solo quiero acotar algo yo tenia el mismo problema en X96 mini (amlogic s905w), porque el equipo funciona con banda de 2,4 GHz. "Mi banda local es de 5GHz así que la tuve que bajar". La tarjeta de red inalámbrica del dispositivo es solo compatible con esa frecuencia. Espero sea eso amigo suerte.

bennington111, 2020-09-07:
> Hola, solo quiero acotar algo yo tenia el mismo problema en X96 mini (amlogic s905w), porque el equipo funciona con banda de 2,4 GHz. "Mi banda local es de 5GHz así que la tuve que bajar". La [...]

I use 2,4 GHz, not 5GHz. But wifi not working.

cponzoni, 2020-09-20:
wifi works again in 3.8 test release

shantigilbert, 2020-11-26:
closing this issue, as it seems to be fixed and no more info has been given, feel free to reopen if that is not the case

closed 2020-11-26 by shantigilbert

alealeks, 2025-06-03:
**EmuELEC V4.7:** WiFi doesn't work on X96 mini. Wi-Fi checkbox is enabled and Wifi hotspot name and password are entered correctly. Everything works correctly on Android.

worstcase-scenario, 2025-06-03:
Did you check if it's perhaps an issue with your password?
Read this: https://github.com/EmuELEC/EmuELEC/wiki/WiFI#important-all-versions

worstcase-scenario, 2025-06-04:
Perhaps you should try and update to 4.8-test (beta) version and see if you have the same issues with that version.

alealeks, 2025-06-04:
> Perhaps you should try and update to 4.8-test (beta) version and see if you have the same issues with that version.

Installed the latest beta version (EmuELEC-Amlogic-ng.aarch64-4.8-TEST-04212024-Generic.img) and the result is the same. Available networks are listed, but it does not connect.