EmuELEC/EmuELEC #263: Menu item in gamelist is missing after quit game

ws02589111, 2020-08-11:
**To Reproduce**
Steps to reproduce the behavior:
1. enter the game
2. press select + start，enter  the quick menu then quit game

Not 100%, but the probability is very high

**Screenshots**
[image]

**EmuELEC version:**
Emuelec v3.7

**Device (please complete the following information):**
S905M2， S905L

shantigilbert, 2020-08-11:
Do yo use CN characters ?

ws02589111, 2020-08-11:
Yes，i replace the font to Microsoft yahei

shantigilbert, 2020-08-11:
Unfortunately that seems to be an issue with ES and CN characters, I do not have a solution.

funwun, 2020-10-28:
same issue~~~

shantigilbert, 2020-11-26:
This has been fixed (or should be), it was a problem with ES vram. But feel free to reopen if its not fixed for you.

closed 2020-11-26 by shantigilbert