EmuELEC/EmuELEC #265: Can't change to 50hz resolution

Cirillone, 2020-08-11:
Hello,

i have a problem when i try to change resolution on Emuelec 3.7. I can only switch to 576p50hz, 1080 and 720 are not working. If i set for 720p50hz on Emuelec, Emulationstation reloads, Emuelec GUI report that the resolution is correct (720p50) but there's only a resolution change to 720p, while the refresh rate stay at 60hz, same happens at 1080.Here the capability of my monitor:

EmuELEC:~ # cat /sys/devices/virtual/amhdmitx/amhdmitx0/disp_cap
480p60hz
576p50hz
720p60hz
1080i60hz
1080p60hz*
720p50hz
1080i50hz
1080p50hz

This problem happens on 3.7 and older 3.6 version. Not tried on older releases.

I tried on 2 different Android boxes on 2 different monitors (1 moniitor and 1 TV both are up to 1080 capable screens).

The andorid boxes are:

1) Sammix R95 1GB RAM S905x
2) Beelink Mini MXIII 2GB RAM S905

I did a clean install on formatted SD.

Let me know if you need further infos. Many thanks for your support!

AlienatedDiego, 2020-08-12:
50hz resolutions have been tested and work correctly in 3.7 and earlier versions. it seems like the problem only occurs to you. check if your screens are indeed compatible with 50hz, do not use an adapter cable or something similar and use clean installation using a different sd with an image from the official emuelec github.

Cirillone, 2020-08-12:
Hi @AlienatedDiego . Thanks for your reply. We discussed this problem a couple of days ago on discord. I confirm that my screens works at 50hz without any problem using any source (CoreElec, Android devices, Blu-Ray...). I confirm also a clean install on different SD with official Emuelec images. Could a wrong dtb file be responsible of this problem? Even if everything else work (Wifi..)? Let me know if there's some log that could be useful to understand better the issue.

AlienatedDiego, 2020-08-12:
well. I have been able to verify that the problem exists and occurs on s905 family devices using non-NG image. The resolutions are correctly displayed at 720p and 1080p but the refresh rates are not shown at 50hz even though they are selected in the EmuELEC GUI, they are only showing at 60hz, as you have said.

In any case, it is not the consequence of an incorrect .dtb.

The developer will be informed in this issue, if it is corrected in future releases or there is news in this matter, you will be informed.

thanks for your report

shantigilbert, 2020-08-12:
If this is the case and its only happening on "Amlogic" project (not on "Amlogic-ng") then I don't think a fix will be coming, the s905 chip ( and the "Amlogic" project itself is EOL) no more updates are made to the base kernel. 

A fix might be possible, but unfortunately it won't be coming from me, as I do not have a 50hz capable screen to even test nor the knowledge to do it.

Cirillone, 2020-08-13:
Ok, understood. Thanks for your support. I could buy a cheap S905x2 or S905x3 later, but i don't understand if this new platform now is really working correctly or not at 50hz on 720 and 1080p resolutions. Someone with a 50hz screen can confirm this? Tnx!

AlienatedDiego, 2020-08-13:
Yes, resolutions with a 50hz refresh rate have been tested and will work fine on NG s905x2, s905x3 and s922x devices.

closed 2020-10-08 by shantigilbert