EmuELEC/EmuELEC #266: Can't get past Dreamcast timeset bios screen on Reicasta_old Core

mymo1982, 2020-08-13:
Hello,
my system is an MXQ Aml S905x 1GB running Emuelec 3.7. Roms are on a dedicated 32 GB SD card.
I configured my 10 button arcade stick, recognized as a generic Dragon joystick, through the Configure Reicast script provided with Emuelec.
When starting a game via Reicasta_old Core, I get the bootup screen with the Dreamcast logo, then the time and date set screen comes up: the joystick lever works fine, I can slide through any fields by pulling right or left and I can also scroll through the date and time figures by pushing it up or down, but when I am finished and I choose the "SELECT" option, no button on my joystick is recognized (no matter how hard or for how long I press them of course), and I cannot get past the screen to play the game. I can get back to Emustation by pressing START+SELECT though.
I can work aroud this issue by using flycast core, which works okay, but I am pretty sure Reicast old is better streamlined for my system. Can anyone help?

shantigilbert, 2020-08-18:
You need a keyboard to set your buttons on this core, please check the "setup category"

mymo1982, 2020-08-18:
> You need a keyboard to set your buttons on this core, please check the "setup category"

Thanks for your answer. 
Of course I used a keyboard to set up my joystick through the Configure reicast script, and all the buttons are recognized by the script, though when I start the core, it only recognizes the lever. Is there any way to skip past the date set bios?

mymo1982, 2020-08-27:
So I guess I solved the issue by correctly configuring the DragonRise Generic Joystick and Reicast Old colre. It only works with games that don't require a DPAD. If someone is running in the same problem, let me know.

closed 2020-08-27 by mymo1982

shantigilbert, 2020-08-27:
For future requests, Can you elaborate on how you configured the joystick?

mymo1982, 2020-08-27:
1) Configure the Joystick via the Script provided with Emuelec.
2) When starting the Script it will ask the device Id you want to use. 
Choose and note down that number associated to DragonRise Generic Usb.
3) Avoid choosing an exit button, Start + Select work just fine in Reicast
4) For some reason, even if the Script recognizes all the buttons, some 
won't be recognized by Reicast, so the only thing left to do is to chosse 
another button. In my case button 295 wouldn't work. It's a trial and error 
procedure.
5) You might want to try and configure the Joystick as both a DPAD and an 
Analog, but that will create a conflict between the two, so I'd only 
configure the analog.
6) Once finisced, open and copy the content of the file 
/storage/.config/reicast/mappings/controller_DragonRiseInc.GenericUSBJoystick.cfg 
(or similar). It's usually not necessary to copy what's written after the 
[Compat] label, except for the entries

bnt_trigger_left
bnt_trigger_right

7) Backup and delete the content of the file 
/storage/.config/reicast/mappings/controller_generic.cfg. Paste into it the 
content of controller_DragonRiseInc.GenericUSBJoystick.cfg that you 
previously copied, save and exit the file.
8) Open the file /storage/.config/reicast/emu_old.cfg and at the end of the 
entry

[input]
evdev_device_id_1 =

add the number you noted down at point n. 2). Save and Exit the file.

I know there are other procedures online but this is what worked for me.

evice_id_1 =