EmuELEC/EmuELEC #268: Error

Frisk-0, 2020-08-15:
I have false positive error message regarding awbios.zip for atomiswave. My games runs perfectly. 

[image]

I also have the error two times for diablo, one at launching and one at exiting but game works great otherwise.

[image]

I'm using gt king s922x and when the error message is displayed, emulationstation runs slow. Have to restart it.

shantigilbert, 2020-08-16:
This are not false positives, please get the proper files as described in the error

closed 2020-08-16 by shantigilbert

TheLastProject, 2020-08-16:
See https://github.com/EmuELEC/EmuELEC/wiki/Common-Questions#why-doesnt-game-x-work

Frisk-0, 2020-08-16:
Thx for pointing me the wiki but this is not relevant here because all my games work.

Why show an error message if game is not crashing and works properly ?

For example, saturn require only one bios and is not region dependant so why show an error for missing bios if every games work with only one ?

Another example, MSX, dozens of useless files are mentionned missing but all my romset works as expected.

This is only frustration to gather useless files only to avoid an "error" message which is not one error because game work.

TheLastProject, 2020-08-16:
It is relevant, the page I linked states exactly what your error means.

TheLastProject, 2020-08-16:
It's also not an error message, it is a warning: "the game **may** not work".

Maybe we can only show this after a crash but using an untested BIOS file can cause other random "weird" issues so I'm not 100% sure if that's the right solution either.

shantigilbert, 2020-08-18:
As for the slowdown, it has been fixed and will be available on the next test release :)