EmuELEC/EmuELEC #270: Some gamepad buttons cannot be recognized

fayehoo, 2020-08-17:
**Describe the bug**
I used the Gamesir gamepad, but the “guide“ (home) button cannot be recognized，this button is recognized as a keyboard.
I noticed that this is not a problem with my own gamepad brand.
But in windows or ubuntu, it's okay.

**To Reproduce**
1. Go to gametroller settings
2. Click on '”guide“' button on gamepad
3.  Dispany information is “Keyboard”
4. The “guide” button can not be recongnized as a button in Emuelec

**Expected behavior**
1.I hope all keys can be recognized and used normally
2.Is it possible to replace the driver, such as xboxdrv, which can recognize the controller compatible with xbox normally.

**Screenshots**
1.run "evtest /dev/input/event5" in terminal ,When pressed "guide" button , the information is:
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x20bc product 0x5500 version 0x101
Input device name: "Xiaoji Gamesir-G3w"
Supported events:
 Event type 0 (Sync)
 Event type 1 (Key)
 Event code 114 (VolumeDown)
 Event code 115 (VolumeUp)
 Event code 116 (Power)
 Event code 128 (Stop)
 Event code 139 (Menu)
 Event code 140 (Calc)
 Event code 142 (Sleep)
 Event code 143 (WakeUp)
 Event code 144 (File)
 Event code 150 (WWW)
 Event code 152 (Coffee)
 Event code 155 (Mail)
 Event code 156 (Bookmarks)
 Event code 158 (Back)
 Event code 159 (Forward)
 Event code 161 (EjectCD)
 Event code 163 (NextSong)
 Event code 164 (PlayPause)
 Event code 165 (PreviousSong)
 Event code 166 (StopCD)
 Event code 167 (Record)
 Event code 168 (Rewind)
 Event code 171 (Config)
 Event code 172 (HomePage)
 Event code 173 (Refresh)
 Event code 217 (Search)
 Event type 4 (Misc)
 Event code 4 (ScanCode)
Testing ... (interrupt to exit)
Event: time 1597646499.253351, type 4 (Misc), code 4 (ScanCode), value c0223
Event: time 1597646499.253351, type 1 (Key), code 172 (HomePage), value 1
Event: time 1597646499.253351, -------------- Report Sync ------------
Event: time 1597646499.397359, type 4 (Misc), code 4 (ScanCode), value c0223
Event: time 1597646499.397359, type 1 (Key), code 172 (HomePage), value 0
Event: time 1597646499.397359, -------------- Report Sync ------------
Event: time 1597646509.709578, type 4 (Misc), code 4 (ScanCode), value c0223
Event: time 1597646509.709578, type 1 (Key), code 172 (HomePage), value 1
2.“172” is the keyboard code of the universal "HOME" key. Is it recognized as a keyboard for this reason?

**EmuELEC version:**
 - Version [3.6]

**Device (please complete the following information):**
 - SOC: [e.g. S905D ]
 - Brand: [e.g. Phicomm ]
 - Model [e.g. N1]
 - Version [e.g. 1.0]
 - Gamepad Brand: [Gamesir-G3W]

**Additional context**
In gamecontrollerdb.txt file, no "guide" on  linux platform ,but in windows platform  for the same gamepad.
For example:
030000005e0400008e02000000007801,ShanWan PS3/PC Wired GamePad,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b10,leftshoulder:b4,leftstick:b8,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b9,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,platform:Windows,
VS
03000000bc2000000055000010010000,ShanWan PS3/PC Wired GamePad,a:b0,b:b1,back:b10,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,leftshoulder:b6,leftstick:b13,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b7,rightstick:b14,righttrigger:a4,rightx:a2,righty:a3,start:b11,x:b3,y:b4,platform:Linux,

shantigilbert, 2020-08-18:
Unfortunately I do not have that problem with any of my controllers, try maybe adding the info to the gamecontrollerdb.txt and see if it does a difference.

Bermarck, 2020-08-29:
You have to configure manually.
First configuration on retroarch and write down wich number is shown in every command. 
Second step, configure manually txt file for controller with number written down for every command.

closed 2020-10-08 by shantigilbert