EmuELEC/EmuELEC #273: Missing key mapping documentation within a game and outside a game

stuart1975, 2020-08-17:
I have an odroid go advance and cannot find any documentation what the buttons can do within a game and outside the game.
for e.g.

 - how to exit a game and automaticlly get back to the game list
 - how to save/load game
    ...

TheLastProject, 2020-08-18:
So, I'm thinking about a "view controls" option under scripts/setup that is automatically launched on first boot, explaining all these controls. What do you think, @shantigilbert?

Do we already have all keys documented on the Wiki or so? Because I honestly don't know them all either

shantigilbert, 2020-08-18:
```
Global hotkey (HK) enable is button V (5)

HK + Dpad up/down changes volume ( I am not sure why the volume works like this, after around 60% you can't barely hear anything on the speakers! this is the same on ALL OGA Images AFAIK!)
HK + Dpad left/right changes brightness
HK + Right bumper toggles between headphones and speakers

Note: That ES does not have a hotkey function so the menu will still move when you set the hotkeys, this also happens in games, just be aware of that!

These hotkeys are NOT the same as the previously posted and those still work!

In-game hotkeys for Retroarch (some of them also work on Amiberry):
Hotkey is button 1 (on OGA v1.1 its III (3))
HK + Up/down controls volume
1 + LB Loads Savestate / RB saves
1 + 4 Shows/hides FPS
1 + 5 fast forward
1 + 6 quits game
1 + x loads menu (this also works with 3 + 4)
1 + b resets
1 + dpad left/right change save slot
For other emulators 1 + 6 quits emulator
```

This migght be a bit different depending on what version of the OGA you have.

stuart1975, 2020-08-18:
> ```
> Global hotkey (HK) enable is button V (5)
>
> HK + Dpad up/down changes volume ( I am not sure why the volume works like this, after around 60% you can't barely hear anything on the speakers! this [...]

Thanks. Please can you add it to the Wiki Page. Because when I searched I could not find it.

This is not right: 
1 + 6 quits game 
I have to use 1 + 2x 6 quits game.

shantigilbert, 2020-08-18:
> I have to use 1 + 2x 6 quits game. 

Only in Retroarch because that is the default setting all the others don't require x2

shantigilbert, 2020-11-26:
Is the information provided enough? I will close this issue but if you feel its not resolved, please reopen