EmuELEC/EmuELEC #282: Pressing Hotkey to quit game.

fayehoo, 2020-08-29:
Now quit the game is "hotkey + start", press twice.  Can emuelec  be set to just press “hotkey” twice to quit the game？It's easier for kids.

shantigilbert, 2020-08-31:
I think you can set it up on Retroarch, but not sure on the other emulators.

Cee123, 2020-10-14:
I was messing around with this a bit the other day. What you can do is go into retroarch, then go into Input > Hotkeys and you can turn off "Confirm Quit" if you want to press the button once to exit a game (i personally find the double press annoying). But anyways, scroll down to the binding for "Restart Retroarch" (quit) and bind that to the same as the home key. But the problem then is because you have a hotkey set and the quit button set as the same as the home button, you cannot use the hotkey for anything else (like accessing the retroarch menu). So also change Menu Toggle Gamepad Combo to L + R together (or any other combo). Then save configuration. So it's pretty easy. You can also set the home button as exit in Drastic.

Cee123, 2020-10-14:
[image]

[image]

[image]

shantigilbert, 2020-11-19:
@fayehoo The solution by @Cee123 is the way to do it. Thanks for sharing.

closed 2020-11-19 by shantigilbert

Cee123, 2020-11-19:
@shantigilbert No worries.

davibaltar, 2022-03-01:
Hello there. I've updated to the version 4.3 and I'm having the same problem. I've tried to configure following the @Cee123 images, but for any hotkey it doesn't work. Does anyone have an idea what it could be?